calyxir/calyx #1589: Lowering to Verilog Broken with Top-Level Component Name Other Than "main"

andrewb1999, 2023-07-08:
The following calyx lowers and simulates correctly:
```
import "primitives/core.futil";
import "primitives/binary_operators.futil";
component main<"toplevel"=1>(@clk clk: 1, @reset reset: 1, @go go: 1) -> (@done done: 1) {
  cells {
    std_slice_0 = std_slice(32, 1);
    @external(1) ext_mem_2 = std_mem_d1(32, 1, 1);
  }
  wires {
    group bb0_8 {
      ext_mem_2.addr0 = 1'd0;
      ext_mem_2.write_data = 32'd7;
      ext_mem_2.write_en = 1'd1;
      bb0_8[done] = ext_mem_2.done;
    }
  }
  control {
      bb0_8;
  }
}
```
whereas the following calyx generates verilog that never finishes simulation:
```
import "primitives/core.futil";
import "primitives/binary_operators.futil";
component dot<"toplevel"=1>(@clk clk: 1, @reset reset: 1, @go go: 1) -> (@done done: 1) {
  cells {
    std_slice_0 = std_slice(32, 1);
    @external(1) ext_mem_2 = std_mem_d1(32, 1, 1);
  }
  wires {
    group bb0_8 {
      ext_mem_2.addr0 = 1'd0;
      ext_mem_2.write_data = 32'd7;
      ext_mem_2.write_en = 1'd1;
      bb0_8[done] = ext_mem_2.done;
    }
  }
  control {
      bb0_8;
  }
}
```
The only difference between these two is the name of the component. Looking at the generated verilog for the second piece of calyx, we can see a main module generated as follows:
```
  module dot(
    input logic clk,
    input logic reset,
    input logic go,
    output logic done
  );
  // COMPONENT START: dot
  string DATA;
  int CODE;
  initial begin
      CODE = $value$plusargs("DATA=%s", DATA);
      $display("DATA (path to meminit files): %s", DATA);
      $readmemh({DATA, "/ext_mem_2.dat"}, ext_mem_2.mem);
  end
  final begin
      $writememh({DATA, "/ext_mem_2.out"}, ext_mem_2.mem);
  end
  ... rest of the dot module ...
  // COMPONENT END: dot
  endmodule

  module main(
    input logic go,
    input logic clk,
    input logic reset,
    output logic done
  );
  // COMPONENT START: main
  wire _guard0 = 1;
  // COMPONENT END: main
  endmodule
```
It seems like the main wrapper for dot is just not generated correctly, leaving a main module that will never assert done.

andrewb1999, 2023-07-08:
If this is too hard to fix now, we should at least fail on programs whose top level module is not named main.

rachitnigam, 2023-07-08:
Hmm, #1491 should have handled this. @andrewb1999 can you check that `wrap-main` is being run and is correctly transforming the program to use the `main` component? `wrap-main` is a part of the default compilation pipeline so it should've worked

rachitnigam, 2023-07-08:
Oh man, that pass just seems broken:
```
% cargo run -- no-wrap.futil -p wrap-main -m file
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s
     Running `target/debug/calyx no-wrap.futil -p wrap-main -m file`
import "primitives/core.futil";
import "primitives/binary_operators.futil";
component dot(@clk clk: 1, @reset reset: 1, @go go: 1) -> (@done done: 1) {
  cells {
    std_slice_0 = std_slice(32, 1);
    @external ext_mem_2 = std_mem_d1(32, 1, 1);
  }
  wires {
    group bb0_8 {
      ext_mem_2.addr0 = 1'd0;
      ext_mem_2.write_data = 32'd7;
      ext_mem_2.write_en = 1'd1;
      bb0_8[done] = ext_mem_2.done;
    }
  }
  control {
    bb0_8;
  }
}
component main<"toplevel"=1>(@go go: 1, @clk clk: 1, @reset reset: 1) -> (@done done: 1) {
  cells {}
  wires {}
  control {}
}
```

closed 2023-07-08 by rachitnigam