calyxir/calyx #1594: MrXL: one-pager is stale

anshumanmohan, 2023-07-10:
The [one-pager](https://docs.calyxir.org/frontends/mrxl.html) about MrXL has a few issues. 

- It starts off with an example that would not actually run all the way (it employs a parallelism factor > 1 on a `reduce`). I get that this is maybe an aspirational example, but in that case we may need to have text in place that explains this fact.
- That example is also perhaps a little too complicated for a first taste of MrXL (e.g., the `map` is mapping over two vectors simultaneously).

I propose we do one of two things:
1. Replace most of the one-pager with the overview section of the [frontend tutorial](https://docs.calyxir.org/tutorial/frontend-tut.html#mrxl-overview). I think that example is neater and tells no untruths. At the end, we could say something like 
    > Intrigued? Confused? Go check out the [frontend tutorial](https://docs.calyxir.org/tutorial/frontend-tut.html) to understand MrXL more deeply and to make it better!
2. Delete the MrXL one-pager entirely, saving MrXL just for the frontend tutorial. Perhaps too drastic!

sampsyo, 2023-08-15:
Good question. One positive about the complex lead example is that it show off all of the language's features, including parallelism and "implicit zips" (mapping over multiple input arrays together in one line), whereas sum-of-squares prioritizes digestibility over showcasing the whole language. In that sense, I don't think it's the _worst_ way to introduce the very broad idea here—via an "end goal" rather than a "first step."

What do you think about doing these two small things?

1. Mentioning in the "compiling to Calyx" section that our implementation doesn't (yet!) support the parallelism from the example.
2. In the top example section, acknowledging that we are not giving a detailed explanation of the language here and linking to the sum-of-squares example section in the tutorial for that.

anshumanmohan, 2023-08-15:
I like it! This is a nice middle path between the status quo and my "tell no untruths" angle, haha. PR coming up tomorrow.

PR #1669 (2023-08-17): MrXL: Update one-pager
This PR updates the MrXL [one-pager](https://docs.calyxir.org/frontends/mrxl.html), clarifying that the example up top cannot actually be compiled to Calyx. It also clarifies the pedagogical goal of the language and provides appropriate links to the frontend tutorial.

Closes #1594.
closed 2023-08-17 by anshumanmohan