calyxir/calyx #1608: Run CIRCT Integration Tests in CI

andrewb1999, 2023-07-16:
We should work towards including the CIRCT frontend flow in CI for integration testing. This way we can know what changes to Calyx may affect the CIRCT frontend and also give us much more confidence in the quality of the CIRCT frontend. This will especially be useful since it will be much easier to create integration test cases than the CIRCT FileCheck test cases.

rachitnigam, 2023-07-16:
Heck yes! This is something we should definitely invest time in. The first step is figuring out if there a pre-built CIRCT/AMC docker we can pull in. If not, we can try setting it up using [Depot](https://depot.dev/blog/introducing-depot) which is what we use for the Calyx docker files.

Once we have that, other things should be pretty straightforward!

sampsyo, 2023-07-17:
Good point about pulling in a pre-built Docker thing; that would make things a lot easier! Barring that, setting up this kind of testing as a nightly workflow (as opposed to the standard GitHub Actions "on every commit to `master` and every pull request") action could help cope with how slow it will be to build the CIRCT stuff from scratch.

andrewb1999, 2023-07-19:
Yeah this definitely seems like something that should be run nightly. I'm not sure there's anywhere that builds docker images for CIRCT with any regularity. Running nightly on the current heads of both CIRCT and Calyx seems like it would work out fine.

rachitnigam, 2023-08-11:
@andrewb1999 since we've punted on the deadline, let's figure out how we can get this working sooner rather than later

rachitnigam, 2023-08-17:
CIRCT binaries are now available as release artifacts: https://github.com/llvm/circt/pull/5832

SpriteOvO, 2023-08-17:
> CIRCT binaries are now available as release artifacts: [llvm/circt#5832](https://github.com/llvm/circt/pull/5832).
> 
> EDIT: Sorry, that PR only uploads the static libs for the binaries. We might need to do some more work to enable building of `hlstool` binary or something.

No, that PR uploads all binaries for both libraries and executables, `hlstool` is also included.

`circt-full-shared-libs-{os}-{arch}.tar.gz` means `circt-full` (bin, lib, headers, also these for llvm) with CMake option `BUILD_SHARED_LIBS` enabled.

rachitnigam, 2023-08-17:
Oh, awesome! I'm glad to be wrong about this then!