calyxir/calyx #1633: FIFO: main loop is inelegant

anshumanmohan, 2023-07-24:
The main loop of the FIFO implemented in Calyx is a little janky. Both of these issues are marked with "AM" in the code, but I'll say more here.

1. If a `pop` fails, the fifo component still sends back some value in the `ans` ref register. We should not write this value to the answer list.
2. After we hit `i=15`, we should break the loop more elegantly than we do now.

The cause of both these issues is the same: we'd like to do something based on the value of the `err` flag, but the guard of the loop itself has an iron grip on `err` and does not want anyone else to look at it.

These issues are both suppressed right now in hacky ways:
1. When the fifo component returns an error, I make it also zero out its `ans` register. So, although the main loop _does_ write an invalid answer to its answer list, that invalid answer is `0` and therefore does not cause undue confusion.
2. I just have a bespoke component `raise_err_if_i_eq_15` whose whole job is to potentially raise the `err` flag. This works fine, it's just overkill and there probably exists a nicer way. This is based on my observation that, although the loop condition likes to maintain a hold on `err`, passing `err` by reference and letting the called component change `err` is still allowed. If we don't like this loophole, we should prevent it.

_Originally posted by @anshumanmohan in https://github.com/cucapra/calyx/issues/1613#issuecomment-1648389544_

rachitnigam, 2023-07-25:
Re: (1) You could consider using the `undef` primitive to write an `'x` into the `ans` memory when the failure occurs to more obviously signal a problem with the underlying design as well. The trade-off is that Verilator doesn't really respect it `'x` propagates into everything.

anshumanmohan, 2023-07-27:
Eventually I would love to do this `undef` thing. However, this can't happen until the fundamental issue is solved: that is, right now I _am_ writing the FIFO component's answer into the user-facing answer memory even after an unsuccessful `pop`. This happens exactly one time, and then the big `while` loop guard correctly ends the loop.

For example, say the main loop's charge is to:
1. push 5
2. pop
3. pop
4. ...

### What I'd like is:

#### FIFO component
On failure of any sort, raise the `err` flag and write `'x` into  the `ans` cell. The rest is as we know it, and is not relevant to the discussion.

#### Main loop
1. Great, push 5. No effect on the user-facing answer memory. It looks like `[0, 0, 0, ...]`.
2. Popped the FIFO and received a down `err` flag and a value of `5` in the `ref ans` cell. Great, write `5` to the user-facing answer memory. It looks like `[5, 0, 0, ...]`.
3. Popped the FIO and received a raised `err` flag and a value of `'x` in the `ref ans` cell. The `'x` doesn't really matter, since I'm clever: I check the `err` flag and know that whatever happened up there should not be committed to the answer memory. It remains `[5, 0, 0, ...]`.
4. While loop checker: sees the `err` flag is up, and quits.

### What I have, sadly, is:

#### FIFO component
On failure of any sort, raise the `err` flag and write `0` into  the `ans` cell. 

#### Main loop
1. Great, push 5. No effect on the user-facing answer memory. It looks like `[0, 0, 0, ...]`.
2. Popped the FIFO and received a down `err` flag and a value of `5` in the `ref ans` cell. Great, write `5` to the user-facing answer memory. It looks like `[5, 0, 0, ...]`.
3. Popped the FIO and received a raised `err` flag and a value of `0` in the `ref ans` cell. However, I have no way of reading the `err` flag since the `while` loop's combinational guard is preventing access to it. So I write what I got, a zero, to the answer at spot 2. It now looks like `[5, 0, 0, ...]`.
4. While loop checker: sees the `err` flag is up, and quits.

So you can see, I've approximated the behavior I want, but I'm relying on the fact that writing `0` looks the same as not writing anything. This is way jankier than I would like. As far as I understand, it all comes from not being able to examine the `err` flag inside the loop body.

anshumanmohan, 2023-08-15:
Just to keep this issue tractable: the code no longer lives under `fifo.py`, but rather under [`queue_call.py`](https://github.com/cucapra/calyx/blob/master/calyx-py/calyx/queue_call.py). The same issues exist, and are still marked with "AM". This common `main` function is used in all three queue programs.

anshumanmohan, 2023-08-15:
@rachitnigam if we do a call to go over the queues' control logic, we should add this to the agenda too!

rachitnigam, 2023-08-16:
Yeah, we could open this call up to the broader Calyx meeting in our first week?

sampsyo, 2023-08-17:
Thanks for the detailed writeup! To summarize what I understand about the problem, the main control loop is approximately:

```
while (not err) and (cmd_idx < MAX_IDX):
  err, val = do_the_queue_stuff(commands[cmd_idx])
  if commands[cmd_idx] is a pop:
    out[out_idx] = val
    out_idx += 1
  cmd_idx += 1
```

And you would like to add this right before the `if`:

```
if err:
  break
```

or, equivalently, just put the entire rest of the loop body under an `if not err`. Is this an accurate summary?

The thing I'm still not quite clear on is: it sounds like you're saying that Calyx doesn't let you read from `err.out` inside the body of the loop, because `err.out` is already used in the condition of the loop. Is that right? If so, I guess I'm surprised… maybe somebody could help me understand what Calyx wellformedness rule this violates? (Since they're both just reads from a stable port, it seems like it should be allowed to read this twice, even concurrently.)

sampsyo, 2023-08-17:
As a separate aside, I don't think undefinedness/`'x` is what we want here. I would argue that putting zero into `ans`, as it does currently, is exactly what the queue should do for non-pops. (You could also make a good argument for "leave it alone.") We as a project still have open questions about what undefinedness should mean, but one thing I'm somewhat confident about is that state elements (memories & registers) should never be undefined. I think it should be UB to write an undefined value into a register.

One reason (among several) is that, in real hardware, there is no "undefined" value. You can't check whether a register is defined, for instance, and take a distinct action for the three cases of 1, 0, or undef. This is a longer and very much separate discussion, but I just wanted to point out that "undefined" is not a good implementation for, like, an `Option` type in application-level semantics. For that, you want a separate bit that indicates `Some` vs. `None`.

anshumanmohan, 2023-08-17:
Yes, that's an accurate summary for the present purposes! (I give that caveat because I don't do the `...and (cmd_idx < MAX_IDX)` as cleanly as I'd like, but let's set that aside for now.)

Your `if err: break` thing would work great, if realizable. I'd also be happy with:

```
while (not err) and (cmd_idx < MAX_IDX):
  err, val = do_the_queue_stuff(commands[cmd_idx])
  if (not err) and (commands[cmd_idx] is a pop): // <-- change is here
    out[out_idx] = val
    out_idx += 1
  cmd_idx += 1
```

I have pushed my wish into a new branch; see [this](https://github.com/cucapra/calyx/commit/5f6fe566ad3bfe8167f40b5e2ece1421f55feda5) commit. 

Here's the error I get:
```
[fud] ERROR: `/scratch/anshuman/calyx/target/debug/calyx -l /scratch/anshuman/calyx -b verilog' failed:
=====STDERR=====
Error: Malformed Structure: Obviously conflicting assignments found:
207 |      err_eq_0.left = err.out;
207 |      err_eq_0.left = err.out;

<stdin>
252 |              if err_eq_0.out with err_eq_0_group {
    |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ Assignments from `err_eq_0_group' are actived here

=====STDOUT=====
```

sampsyo, 2023-08-19:
Thanks for clarifying! So the wellformedness checker is reflecting a policy that says "the `with` comb-group may be active during the entire execution of the `while` body."  This is (yet another) situation in which we have some fuzzy *de facto* implementation of a correctness constraint but it is not fully codified: it is not precisely spelled out [in the docs][ub], nor do we have a dynamic checker to enforce it. It would be great to more completely cement this entire category of stuff someday.

Anyway, for the time being, the workaround would be to put that comparison into a register. That is, add a 1-bit register, write the value of `err_eq_0.out` into it at some point in the body, and use `my_great_new_reg.out` for both conditions. That way, neither condition needs to write to anything; they both just read from the register, and concurrent reads are allowed.

[ub]: https://docs.calyxir.org/lang/undefined.html

PR #1681 (2023-08-21): Builder: use register to make the queue nicer
The main while loop in all the queue programs was a little inelegant: the combinational group that guarded the main while loop was preventing access to a key port within the loop.

This PR employs a register to avoid the issue of a combinational group. The resultant code is better in two ways:
- It no longer needs a whole separate component just to raise a flag in case of `i == 15`.
- In case the queue returns an `err`, the main loop correctly simulates a `break` immediately, instead of stumbling through the rest of the body and then getting broken [... cut]
closed 2023-08-21 by anshumanmohan