calyxir/calyx #1640: Comb components not handled properly during clk and reset insertion passes

andrewb1999, 2023-07-26:
I'm running into an issue where comb components are expected to have clk and reset ports during the clk and reset insertion passes. If I manually add @clk and @reset ports to the comb components everything works fine.

Here is the calyx code that causes the error (sorry for the length, I haven't had a change to make a shorter example):
```
import "primitives/core.futil";
import "primitives/binary_operators.futil";
import "primitives/memories.futil";
import "primitives/pipelined.futil";
comb component layout_hw0(reform_port_addr0: 2, reform_port_addr1: 2) -> (flat_port_addr0: 4) {
  cells {
    add_0 = std_add(4);
    d1_times_4 = std_lsh(4);
    pad_reform_port_addr1 = std_pad(2, 4);
    pad_reform_port_addr0 = std_pad(2, 4);
  }
  wires {
    flat_port_addr0 = add_0.out;
    add_0.left = pad_reform_port_addr0.out;
    add_0.right = d1_times_4.out;
    d1_times_4.left = pad_reform_port_addr1.out;
    d1_times_4.right = 4'd2;
    pad_reform_port_addr1.in = reform_port_addr1;
    pad_reform_port_addr0.in = reform_port_addr0;
  }
}
comb component layout_hw1(reform_port_addr0: 2, reform_port_addr1: 2) -> (flat_port_addr0: 4) {
  cells {
    add_0 = std_add(4);
    d1_times_4 = std_lsh(4);
    pad_reform_port_addr1 = std_pad(2, 4);
    pad_reform_port_addr0 = std_pad(2, 4);
  }
  wires {
    flat_port_addr0 = add_0.out;
    add_0.left = pad_reform_port_addr0.out;
    add_0.right = d1_times_4.out;
    d1_times_4.left = pad_reform_port_addr1.out;
    d1_times_4.right = 4'd2;
    pad_reform_port_addr1.in = reform_port_addr1;
    pad_reform_port_addr0.in = reform_port_addr0;
  }
}
component mem_0_comp(@clk clk: 1, port0_addr0: 2, port0_addr1: 2, @go @static(1) port0_read_en: 1, port1_addr0: 2, port1_addr1: 2, port1_write_data: 32, @go(2) @static(1) port1_write_en: 1, @reset reset: 1) -> (@stable port0_read_data: 32, @done port0_read_done: 1, @done(2) port1_write_done: 1) {
  cells {
    layout_hw_inst1 = layout_hw1();
    layout_hw_inst0 = layout_hw0();
    mem_0_comp_bank_0 = ram_1r1w(32, 16, 4, 16, 4);
  }
  wires {
    mem_0_comp_bank_0.clk = clk;
    layout_hw_inst1.reform_port_addr0 = port1_addr0;
    layout_hw_inst1.reform_port_addr1 = port1_addr1;
    layout_hw_inst0.reform_port_addr0 = port0_addr0;
    layout_hw_inst0.reform_port_addr1 = port0_addr1;
    port0_read_data = mem_0_comp_bank_0.port0_read_data;
    mem_0_comp_bank_0.port0_read_en = port0_read_en;
    port0_read_done = mem_0_comp_bank_0.port0_read_done;
    mem_0_comp_bank_0.port0_addr0 = layout_hw_inst0.flat_port_addr0;
    mem_0_comp_bank_0.port1_write_data = port1_write_data;
    mem_0_comp_bank_0.port1_write_en = port1_write_en;
    port1_write_done = mem_0_comp_bank_0.port1_write_done;
    mem_0_comp_bank_0.port1_addr0 = layout_hw_inst1.flat_port_addr0;
  }
  control {
  }
}
extern "ram_1r1w.sv" {
  primitive ram_1r1w[WIDTH, PORT0_SIZE, PORT0_IDX_SIZE, PORT1_SIZE, PORT1_IDX_SIZE](@clk clk: 1, port0_addr0: PORT0_IDX_SIZE, @go @static(1) port0_read_en: 1, port1_addr0: PORT1_IDX_SIZE, port1_write_data: WIDTH, @go(2) @static(1) port1_write_en: 1) -> (@stable port0_read_data: WIDTH, @done port0_read_done: 1, @done(2) port1_write_done: 1);
}
component mem_1_comp(@clk clk: 1, port0_addr0: 2, @go @static(1) port0_read_en: 1, port1_addr0: 2, port1_write_data: 32, @go(2) @static(1) port1_write_en: 1, @reset reset: 1) -> (@stable port0_read_data: 32, @done port0_read_done: 1, @done(2) port1_write_done: 1) {
[... 551 lines cut ...]
                }
              }
            }
          }
          phase_reg_19;
        }
      }
      static par {
        cond_init_5;
        assign_loop_6_init_0;
      }
      while loop_6_cond_reg.out {
        seq {
          load_5;
          par {
            cond_5;
            store_4;
            phase_reg_21;
          }
        }
      }
    }
  }
}
```

rachitnigam, 2023-07-26:
Thanks for opening the issue! Should we allow `comb` components to have `@clk` and `@reset` signals? I don't think there is any harm in doing it but also maybe we want to enforce that you can't use clocked primitives in `comb component`s?

andrewb1999, 2023-07-26:
I guess there's no harm as long as we enforce that you can't use clocked primitives.

rachitnigam, 2023-07-26:
Done in the PR!

reset, 2023-07-26:
Great work here boys, I approve this PR 👍

closed 2023-07-26 by rachitnigam

desc, 2023-07-27:
> Great work here boys, I approve this PR 👍

I can totally relate to this. 👍