calyxir/calyx #1645: Overly aggressive dead assignment removal

bcarlet, 2023-08-04:
When compiling the following component, the dead assignment removal pass completely removes the assignments to `rsh.left` and `rsh.right`, leading to undriven ports:

```
import "primitives/core.futil";

component high(x: 16) -> (out: 8) {
  cells {
    r = std_reg(8);
    rsh = std_rsh(16);
    slice = std_slice(16, 8);
  }
  wires {
    group write {
      rsh.left = x;
      rsh.right = 16'd8;
      r.in = slice.out;
      r.write_en = 1'd1;
      write[done] = r.done;
    }
    slice.in = rsh.out;
    out = r.out;
  }
  control {
    write;
  }
}
```

This code seems legal to me; it runs fine in the interpreter, and in the simulator when the dead assignment removal pass is disabled, so it seems like this is a bug in the pass.

rachitnigam, 2023-08-04:
This pass is the bane of my existence

rachitnigam, 2023-08-04:
Oh wait, this program is a little funky: the continuous assignments are only meaningful when the group is active and vice versa. I guess one possible problem could be that when removing the writes to `rsh`, we're not considering the continuous read from `rsh.out`.

EclecticGriffin, 2023-08-07:
Independent of the probably over-zealous pass, I'd argue that this program/component has undefined behavior as `rsh.out` is only driven when the group is running. At all other times, say in-between component executions or after the done port is high, the output is no longer valid.

rachitnigam, 2023-08-10:
@EclecticGriffin not quite: the `out` port is connected to the register which has a stable value. I agree the program is a bit tricky though because the group's execution uses the continuous assignments to update its value but not sure if this is undefined behavior.

closed 2023-08-10 by rachitnigam

rachitnigam, 2023-08-10:
I think you're right that if one of those cells was not a stable one, it would definitely be undefined but this example is a little tricky!

PR #1652 (2023-08-10): `dead-assign-removal` considers continuous assignments
Fixes #1645