calyxir/calyx #1646: comb-prop not propagating value to some consumers

matth2k, 2023-08-07:
```
import "primitives/core.futil";
component amcMemory0_comp(@clk clk: 1, port0_addr0: 10, @go @static(1) port0_read_en: 1, port1_addr0: 10, port1_write_data: 32, @go(2) @static(1) port1_write_en: 1, @reset reset: 1) -> (@stable port0_read_data: 32, @done port0_read_done: 1, @done(2) port1_write_done: 1) {
  cells {
    merge_hw_3_port_2_write_en_gated = std_and(1);
    merge_hw_3_port_1_write_en_gated = std_and(1);
    merge_hw_3_write_dones_or_d1_i0 = std_or(1);
    merge_hw_3_write_done_port1 = std_wire(1);
    merge_hw_3_write_done_port2 = std_wire(1);
    merge_hw_inst3 = merge_hw();
    merge_hw_2_port_2_read_en_gated = std_and(1);
    merge_hw_2_port_1_read_en_gated = std_and(1);
    merge_hw_2_data_latch_predicate_false = std_not(1);
    merge_hw_2_data_latch_new_value = std_wire(32);
    merge_hw_2_data_latch_module_value = std_wire(32);
    merge_hw_2_data_latch_predicate_true_d1_i0 = std_or(1);
    merge_hw_2_data_latch = std_reg(32);
    merge_hw_2_read_dones_or_d1_i0 = std_or(1);
    merge_hw_2_read_done_port1 = std_wire(1);
    merge_hw_2_read_data_port1 = std_wire(32);
    merge_hw_2_read_done_port2 = std_wire(1);
    merge_hw_2_read_data_port2 = std_wire(32);
    merge_hw_inst2 = merge_hw();
    amcMemory0_comp_bank_0 = ram_1r1w(32, 512, 9, 512, 9);
    amcMemory0_comp_bank_1 = ram_1r1w(32, 512, 9, 512, 9);
  }
  wires {
    merge_hw_2_data_latch.clk = clk;
    amcMemory0_comp_bank_0.clk = clk;
    amcMemory0_comp_bank_1.clk = clk;
    merge_hw_3_port_2_write_en_gated.left = merge_hw_inst3.port2_select;
    merge_hw_3_port_2_write_en_gated.right = port1_write_en;
    merge_hw_3_port_1_write_en_gated.left = merge_hw_inst3.port1_select;
    merge_hw_3_port_1_write_en_gated.right = port1_write_en;
    port1_write_done = merge_hw_3_write_dones_or_d1_i0.out;
    merge_hw_3_write_dones_or_d1_i0.left = merge_hw_3_write_done_port1.out;
    merge_hw_3_write_dones_or_d1_i0.right = merge_hw_3_write_done_port2.out;
    merge_hw_inst3.merged_addr0 = port1_addr0;
    merge_hw_2_port_2_read_en_gated.left = merge_hw_inst2.port2_select;
    merge_hw_2_port_2_read_en_gated.right = port0_read_en;
    merge_hw_2_port_1_read_en_gated.left = merge_hw_inst2.port1_select;
    merge_hw_2_port_1_read_en_gated.right = port0_read_en;
    port0_read_data = merge_hw_2_data_latch_module_value.out;
    merge_hw_2_data_latch_predicate_false.in = merge_hw_2_data_latch_predicate_true_d1_i0.out;
    merge_hw_2_data_latch.write_en = merge_hw_2_data_latch_predicate_true_d1_i0.out;
    merge_hw_2_data_latch.in = merge_hw_2_data_latch_new_value.out;
    merge_hw_2_data_latch_module_value.in = merge_hw_2_data_latch_predicate_true_d1_i0.out ? merge_hw_2_data_latch_new_value.out;
    merge_hw_2_data_latch_module_value.in = merge_hw_2_data_latch_predicate_false.out ? merge_hw_2_data_latch.out;
    merge_hw_2_data_latch_new_value.in = merge_hw_2_read_done_port1.out ? merge_hw_2_read_data_port1.out;
    merge_hw_2_data_latch_new_value.in = merge_hw_2_read_done_port2.out ? merge_hw_2_read_data_port2.out;
    merge_hw_2_data_latch_predicate_true_d1_i0.left = merge_hw_2_read_done_port1.out;
    merge_hw_2_data_latch_predicate_true_d1_i0.right = merge_hw_2_read_done_port2.out;
    port0_read_done = merge_hw_2_read_dones_or_d1_i0.out;
    merge_hw_2_read_dones_or_d1_i0.left = merge_hw_2_read_done_port1.out;
    merge_hw_2_read_dones_or_d1_i0.right = merge_hw_2_read_done_port2.out;
    merge_hw_inst2.merged_addr0 = port0_addr0;
    merge_hw_2_read_data_port2.in = amcMemory0_comp_bank_0.port0_read_data;
    amcMemory0_comp_bank_0.port0_read_en = merge_hw_2_port_2_read_en_gated.out;
    merge_hw_2_read_done_port2.in = amcMemory0_comp_bank_0.port0_read_done;
    amcMemory0_comp_bank_0.port0_addr0 = merge_hw_inst2.shared_addr0;
    amcMemory0_comp_bank_0.port1_write_data = port1_write_data;
    amcMemory0_comp_bank_0.port1_write_en = merge_hw_3_port_2_write_en_gated.out;
    merge_hw_3_write_done_port2.in = amcMemory0_comp_bank_0.port1_write_done;
    amcMemory0_comp_bank_0.port1_addr0 = merge_hw_inst3.shared_addr0;
    merge_hw_2_read_data_port1.in = amcMemory0_comp_bank_1.port0_read_data;
    amcMemory0_comp_bank_1.port0_read_en = merge_hw_2_port_1_read_en_gated.out;
    merge_hw_2_read_done_port1.in = amcMemory0_comp_bank_1.port0_read_done;
    amcMemory0_comp_bank_1.port0_addr0 = merge_hw_inst2.shared_addr0;
    amcMemory0_comp_bank_1.port1_write_data = port1_write_data;
    amcMemory0_comp_bank_1.port1_write_en = merge_hw_3_port_1_write_en_gated.out;
    merge_hw_3_write_done_port1.in = amcMemory0_comp_bank_1.port1_write_done;
    amcMemory0_comp_bank_1.port1_addr0 = merge_hw_inst3.shared_addr0;
  }
  control {
  }
}
extern "ram_1r1w.sv" {
  primitive ram_1r1w[WIDTH, PORT0_SIZE, PORT0_IDX_SIZE, PORT1_SIZE, PORT1_IDX_SIZE](@clk clk: 1, port0_addr0: PORT0_IDX_SIZE, @go @static(1) port0_read_en: 1, port1_addr0: PORT1_IDX_SIZE, port1_write_data: WIDTH, @go(2) @static(1) port1_write_en: 1) -> (@stable port0_read_data: WIDTH, @done port0_read_done: 1, @done(2) port1_write_done: 1);
}
comb component merge_hw(merged_addr0: 10) -> (shared_addr0: 9, port1_select: 1, port2_select: 1) {
  cells {
    port2_select_eq = std_eq(1);
    port1_select_eq = std_eq(1);
    shared_addr_slice = std_slice(10, 9);
    select_bits_slice = std_slice(10, 1);
    select_bits = std_rsh(10);
  }
  wires {
    port2_select_eq.left = select_bits_slice.out;
    port2_select_eq.right = 1'd1;
    port2_select = port2_select_eq.out;
    port1_select_eq.left = select_bits_slice.out;
    port1_select_eq.right = 1'd0;
    port1_select = port1_select_eq.out;
    shared_addr_slice.in = merged_addr0;
    shared_addr0 = shared_addr_slice.out;
    select_bits_slice.in = select_bits.out;
    select_bits.right = 10'd9;
    select_bits.left = merged_addr0;
  }
}
```

comb-prop is optimizing away the wire cell `merge_hw_2_data_latch_new_value` and re-anchors the two conditionals assignment directly to register cell input `merge_hw_2_data_latch.in`. However, `merge_hw_2_data_latch_module_value` was another consumer of those same conditional assignments. So it appears to be losing its conditional assignments.

Ultimately,  `merge_hw_2_data_latch_module_value` has 3 cases when you propagate the logic:
1. `(a | b) & a = a` 
2. `(a | b) & b = b`
3. `!(a | b) = !a & !b` (no propagation)

So if the conditional assignment gets delete for `merge_hw_2_data_latch_new_value`, cases 1 & 2 will presumably get deleted as well. I'm having trouble simplifying the test case.

Eventually, I will rewrite this logic using groups, but for now I will have to run `-d comb-prop`.

rachitnigam, 2023-08-11:
@matth2k when you describe the three cases for propagation, what are `a` and `b` supposed to be?

rachitnigam, 2023-08-11:
Here is a smaller test case although not sure if is is exactly the same problem you were running into @matth2k:
```
import "primitives/compile.futil";
component main<"toplevel"=1>() -> (out: 32) {
  cells {
    opt = std_wire(32);
    r = std_reg(32);
    // Stable conditions
    r0 = std_reg(1);
    r1 = std_reg(1);
  }
  wires {
    r.in = opt.out;
    r.write_en = 1'd1;

    opt.in = r0.out ? 32'd10;
    opt.in = r1.out ? 32'd20;

    out = opt.out;
  }
  control {}
}
```

Running `comb-prop` on it, we get:
```
% cargo run -q -- prop.futil -p comb-prop -p dead-cell-removal -m file            
import "primitives/compile.futil";
component main<"toplevel"=1>(@go go: 1, @clk clk: 1, @reset reset: 1) -> (out: 32, @done done: 1) {
  cells {
    opt = std_wire(32);
    r0 = std_reg(1);
    r1 = std_reg(1);
  }
  wires {
    opt.in = r0.out ? 32'd10;
    opt.in = r1.out ? 32'd20;
    out = opt.out;
  }
  control {}
}
```

Note that there is an assignment that uses the output from `opt.out` but there are no assignments to `opt.in` left in the program. Here is the sequence of events that occur in `comb-prop`:
1. Observe `r.in = opt.out` assignment and add the rewrite `opt.in` -> `r.in`.
2. Mutate the assignments with `opt.in` to use `r.in` instead.

Step (2) is the problem here: because we mutate the assignments, `opt.in` is left undriven which means all uses of `opt.out` are wrong now.

The fix is a little tricky: we need to change the pass to only apply mutating rewrites to ports in the source and guards of the assignments. When the `dst` needs to be rewritten, we need to do that by creating duplicate assignment. In case `opt.out` is unused in the program, a future pass, like `dead-assign-removal` should get rid of those assignments.

rachitnigam, 2023-08-11:
A different solution that might be simpler in the short-term but has less optimizing power overall is to simply disable the rewrite `opt.in` -> `r.in` if we ever see `opt.out` being used as a source in the program. This is true of the above program. Overall, however, I feel that we should think more carefully about how `comb-prop` should work from first principles.

PR #1656 (2023-08-11): `comb-prop`: Disable rewrites when wire output is used
Fixes #1646.
closed 2023-08-11 by rachitnigam