calyxir/calyx #1648: `ref` cells: bug in `compile-ref` pass

anshumanmohan, 2023-08-09:
Tinkering with PIFO example (#1625) has revealed a bug in the `compile-ref` pass! Many thanks to @EclecticGriffin for helping me learn how to use the debugger; that was key to this discovery.

Note: Below I'll say `ref-invoke` when I mean an `invoke` with at least one cell passed by reference.

The issue comes up when a `ref-invoke`d component itself `ref-invoke`s a component. I have pushed a minimal breaking example in the branch `ref-bug`, and the links below point to that branch. 

The goal is simple:
- Component [`main`](https://github.com/cucapra/calyx/blob/dab092467726bf8b81f5aecc86919724d9a478be/calyx-py/test/correctness/mbe.py#L32-L40) creates a register called `value` and `ref-invoke`s its component [`incr`](https://github.com/cucapra/calyx/blob/dab092467726bf8b81f5aecc86919724d9a478be/calyx-py/test/correctness/mbe.py#L19-L29) by passing `value` by reference.
- Component `incr` does not do the work itself. It has a cell of its own – the component [`incr_helper`](https://github.com/cucapra/calyx/blob/dab092467726bf8b81f5aecc86919724d9a478be/calyx-py/test/correctness/mbe.py#L6-L16) – which it `ref-invoke`s, again passing `value`.
- `incr_helper` actually sets up a little bit of wiring to increment `value` by one.

Let us generate `.futil` code: 
```
python3 mbe.py > mbe.futil
```

Examine the [resultant Calyx code](https://github.com/cucapra/calyx/blob/ref-bug/calyx-py/test/correctness/mbe.futil), paying attention to the `invoke` statements. So far, so good.

Now let's compile away the `ref`s:
``` 
fud e --to interpreter -s calyx.flags '-p compile-ref' mbe.futil > mbe_sans_ref.futil
```

Now examine the [Calyx code without `ref`s](https://github.com/cucapra/calyx/blob/ref-bug/calyx-py/test/correctness/mbe_sans_ref.futil).

The first `invoke`, where `main` calls `incr`, looks okay: https://github.com/cucapra/calyx/blob/dab092467726bf8b81f5aecc86919724d9a478be/calyx-py/test/correctness/mbe_sans_ref.futil#L80-L89

But the nested `invoke`, where `incr` calls `incr_helper`, has some serious problems:

https://github.com/cucapra/calyx/blob/dab092467726bf8b81f5aecc86919724d9a478be/calyx-py/test/correctness/mbe_sans_ref.futil#L97-L101

The previous examples of `ref` in action have gotten relatively complex, but have not required nesting of this form. Now, when it comes to PIFOs, we do: `main` `ref-invoke`s its `pifo` component, and `pifo` `ref-invoke`s one of its `fifo` components.

Any thoughts appreciated!

rachitnigam, 2023-08-10:
Oh, that's quite weird! Another thing to note here is that `incr_helper` doesn't even get the signature of `value` inlined into its signature which means that for some reason, the `value` ref cell is being completely ignored.

rachitnigam, 2023-08-10:
Ah, actually, I'm wrong: the pass *does* inline the ports of the `ref value` in `incr_helper` but does not actually update the `invoke` to build all the assignments.

rachitnigam, 2023-08-10:
Okay, I've identified the problem and it's mostly because we mutate a bunch information when we're compiling refs which means that certain information needed for correctly generating the assignments to the `invoke` is clobbered before we use it.

rachitnigam, 2023-08-10:
@anshumanmohan how important is this feature in getting the PIFO example working? Unfortunately I'm running into a bunch of bad decisions we made when implementing `compile-ref` and getting to the point where doing what #1649 suggests is probably a better solution but requires more work.

anshumanmohan, 2023-08-10:
Thanks for taking a look, and for the attempted fix in #1650. I see that you have inlined my minimal breaking example as a test case in that PR, so if that works, this issue is indeed closed. After #1650 is merged, I'll try the PIFO example to see if anything else is broken in my code.

> How important is this feature in getting the PIFO example working? 

The way my PIFO program is designed at present, this chain of `ref-invoke`s (`main` --> `pifo` --> `fifo_n`) is necessary. I suppose I could move things around and make it so that all the `ref-invoke`s  are deployed from the main function, such that no chains of length > 1 are needed anymore. This would involve a lot of back and forth between the components `main`, `pifo`, and `fifo_n` in a sort of forced, weird way.

closed 2023-08-10 by rachitnigam