calyxir/calyx #1672: Ship all primitive with the `calyx` binary

rachitnigam, 2023-08-18:
It's kind of annoying that `calyx` alone isn't sufficient to compile arbitrary programs that use known primitives. We should just `include_str!(...)` all the primitives in the `primitive/` folder with the `calyx` binary so people can use it without needing to get the repo source.

This will be relevant for a seamless flow with https://github.com/llvm/circt/pull/5879.

sampsyo, 2023-08-19:
This is definitely a pain point, even outside of CIRCT—it is annoying that our "build products" are insufficient to actually compile stuff. Consequences include:

* it does not suffice to `cargo install calyx` to get a working compiler
* using our current you-must-check-out-the-repo workflow, occasional frustration with configuring fud to point to the right place to find the libraries
* downstream dependents of Calyx cannot just depend on the crate; they have to use a subrepo or something like that

A one-off `include_str!` of the necessary libraries is certainly an option. It has a few downsides, including that you'd need to recompile the compiler to update the libraries (although [a crate like this](https://github.com/magiclen/lazy-static-include) could help), and we'd be "blessing" a certain set of libraries as included (maybe that's fine, if we ever pursue #419 to support other libraries, but if we do that, then the stdlib could instead be "just another library" that is included by default). Perhaps this is the best way! I think "how to ship the standard library with the compiler in a convenient way" is kind of an annoying thing that any compiler has to deal with, so maybe it would behoove us to look/ask around to see if anyone has any particularly clever opinions about the least-bad strategy…

rachitnigam, 2023-08-20:
I think the pain points are well put and need to be addressed to truly have an out of the box working experience. I agree with you that `include_str!` is pretty bad UX and I've outlined some problems in #1678 as well. 

I think real compiler like `rustc` have the benefit of relying on `cargo` and `rustup` to install these tools in known locations and use that as a building off point. We can do something similar as well if we try using a build script of `calyx` but I'm not a super expert in those. 

Regardless, it's again one of those things where a short term solution might look very different from the long term solution.

rachitnigam, 2023-08-20:
Okay, I just [had a good chat](https://chat.openai.com/share/ae482b44-0477-444e-ab7c-93bb9b4fde72) and seems like one possible way to do this is:
1. `include_str!` all the primitives in a new package.
2. Publish that package.
3. Define a new `build.rs` script for the `calyx` crate to depend on this primitives crate.
4. During the build process, materialize the files and write them to the outputs from the build script
5. Move these generated files to the AppData equivalent for the file system.
6. In the `calyx` crate, search the AppData folder to see if the primitives directory was correctly loaded. If it was, use that primitives directory.

This is still pretty baroque but it gives us a bit of flexibility since the interface to the compiler still uses the `-l` flag to specify the primitives folder but additionally gets to use a default installation location when present.

rachitnigam, 2023-08-22:
Unfortunately, #1678 did not fix this because `OUT_DIR` seems like a build time, temporary library.

rachitnigam, 2023-08-23:
Woo! Fixed by #1687.

PR #1678 (2023-08-22): Include the primitive implements with `calyx` binary
Fixes #1672.

This will require some significant rework of the frontend infrastructure to bypass the reading of primitives. We should also figure out how we want the UX of the primitive library to be going forward: do we default to using the primitives in the compiled binary or do we use the one from the file path?
closed 2023-08-23 by rachitnigam