calyxir/calyx #1679: Migrate to new organization

rachitnigam, 2023-08-19:
Calyx is big enough at this point to graduate into its own organization. There is a couple of benefits of being a part of our own organization:
- We can have our own governance and code access model separate from CAPRA. Currently all CAPRA members have write access to this repo.
- We can have an umbrella for various spin-off projects in the Calyx ecosystem that have more dependencies than just the compiler.
  - Specifically, I think `fud` should be moved into its own repository and versioned separately from Calyx. It is a dependency for several tools and it makes less sense to keep it in the repo.
- Calyx can have its own identity separate beyond a research project in CAPRA. Of course, everyone working on Calyx will be grandfathered into the new organization.

I own the [calyxir](https://github.com/calyxir) organization on Github so I'd propose moving the project there. However, I want to have consensus on this decision since it affects a lot of people.

EclecticGriffin, 2023-09-25:
Makes sense to me. What would the process look like?

rachitnigam, 2023-09-26:
Good question but the unfortunate answer is "we just move it and see what breaks". The things that are most likely to break are deployments and the website (since they rely on forwarding that we've setup on the domain manager). 

Apart from that, we'd need to update the docs and other materials to point to the new organization.

sampsyo, 2023-09-26:
One logistical thing I'd recommend, if we do this: we do #1506 (rename the default branch to `main`) at the same time, since (in my experience doing this for other repos) that also involves quite a bit of "just move it and see what breaks" so we might as well do this is one big kerfuffle.

EclecticGriffin, 2023-09-27:
I'm game, is there a best practice for when to do this or do we just let 'em rip and see?

rachitnigam, 2023-10-03:
@sampsyo Agreed! I think we should pick one of the breaks to do this so that it doesn't get in the way of people? Fall break is too close and Thanksgiving is right before the ASPLOS deadline so maybe we delay it to December?

rachitnigam, 2023-12-23:
done

closed 2023-12-23 by rachitnigam