calyxir/calyx #1697: Next Steps for PIFO Trees

calebmkim, 2023-08-25:
(Writing this after a synchronous meeting w/ @anshumanmohan so feel free to add to/correct this post). 

## Support different popping policies 
Instead of having a round robin "pop" policy for PIFO tree, PIFO trees should be able to support a different split (e.g., 70/30 or something like that). 

## PIFO tree can communicate with Static "parent" to determine its popping policy
This was discussed briefly on Monday: we want PIFO trees to communicate with a (parent that can get packet information from the PIFO tree(s) and then do some (hopefully static) calculation, and then send information back to the PIFO tree to update its popping policy (e.g., go from 70/30 to round robin or something like that). 
Drawing is below. We'll start with a parent and just one PIFO tree child, but could possibly add more since this is more realistic iiuc. 
Hopefully, the parent's calculations are simple enough to be static; we would need this to talk about the "mixed static dynamic" story for the paper. 
[image]

One interesting note is that **this is similar to the systolic array stuff I've been doing**. 
In both cases, we have one static component (the systolic array/Parent) and a dynamic component (the post op component/PIFO tree). And in both cases, we need to run them both kind of in parallel[^1] and they need to communicate with each other. I think I have a general idea of how to do this in the Systolic Array case; hopefully doing this with systolic arrays will make it easier to do with PIFO trees? 

[^1]: Systolic Array & Post op need to run in parallel since we want to start the post op on a PE as soon as the given Data is ready (which means some of parts of the systolic array are still calculating) 

## Concurrent pushes and pops 
This needs more thinking. For example, what happens if we pop while pushing something that would have ended up being the head of the PIFO tree? 
Another thing: if I understood correctly, @anshumanmohan's PIFO tree paper didn't allow concurrent pushes/pops, so maybe for now we can focus on the first two things, since they're more concrete? 

If anyone wants to elaborate or correct anything I said, lmk.

sampsyo, 2023-08-28:
> For example, what happens if we pop while pushing something that would have ended up being the head of the PIFO tree?

This is a good question, and one possible resolution is to say that there is _no_ specific latency guarantee about when a push becomes visible to a pop. That is, the only guarantee we'd provide for this concurrent version would be like "the consumer pops values in the order determined by the pushes and the policy for reordering things while in flight"... that is, just about the _order of arrival_, not about the _timing at which the values arrive_.

This would mean, for example, that it would be legal (in a plain FIFO) for the producer to push 5 items in a row over 5 cycles, but a consumer who tries to pop on every cycle during those same 5 cycles to see nothing. Then, 10,000 cycles later, the consumer finally gets the first item that was pushed. Then they see nothing for 7 more cycles and get the second item. Maybe @anshumanmohan knows whether this timing looseness is, like, unworkable for some application reason?

A consequence for the implementation would be that, if there is a push and a pop in the same cycle, then this doesn't need to be handled any nontrivial-depth queue. All items take a while to propagate through the structure, so there is no "within-a-cycle" forwarding necessary.

rachitnigam, 2023-08-29:
I'm not as familiar with the exact networking setup but one thing I can imagine being a problem: if the PIFO tree is used only for guaranteeing some sort of fairness or priority, it might make more sense to allow it to violate that fairness if it means that it can keep producing packets during those 10,000 cycles instead of just blocking.

sampsyo, 2023-08-29:
Sure? I guess my intuition, however, would be that any network setup would care a lot more about bandwidth that latency.

anshumanmohan, 2023-09-12:
Sorry that I'm picking up this thread after so long, all. I think Caleb's summary is accurate.

Adrian brings up an interesting matter of order vs timing. In my understand it _is_ order that matters and not latency (so, said simply, I think Adrian is right) but of course things get more complicated as you go from simpler kinds of queues to more complicated kinds of queues. More about that in a worked example below.

Rachit is bringing up the matter of work conservation. 
- Work-conserving scheduling regimes will never waste an opportunity to send a packet, even a very low-ranked packet, if the downstream network has the ability to pop a packet. 
- Non-work-conserving regimes are much more seriously concerned with fairness: they may waste the downstream network's capacity, keeping a packet in the scheduling queue, if the packet is not "ready" for some reason. Think of this notion of readiness as a "do not pop me before time $t_j$" tag that is attached to the packet when it is enqueued. That is, if it is time $t_i$ and $t_i < t_j$, the tree cannot release that packet.

My OOPSLA work handles the former case only, though I can say more about the latter case.

To hash out Adrian's suggestion some more, here's a worked example using a PIFO or PIFO tree, and not just a FIFO. I've annotated packets with indices and their priorities. There are just two priorities, and let's say it's "first come, first served" service within a priority.

```
cycle 1: push pkt 1 (lo)
cycle 2: push pkt 2 (lo)
...
cycle 10: push pkt 10 (lo)
...
cycle 10,000: push pkt 11 (hi)
cycle 10,001: pop
```

In a perfect world, with packets being propagated into their correct spot in the queue instantly, we would expect `pkt 11 (hi)` to be `pop`ped. This won't actually happen, since we aren't in a perfect world. So how do we get a handle on the semantics of `pop`?

I posit we'll need some notion of _housekeeping_. That is, after a push, it takes a few cycles for the scheduler to be done with all the necessary bookkeeping and to arrive back at a well-formed tree. When that is done, the tree is known to be housekept _up to some command_.

When you `pop`, you are guaranteed the highest-ranked item from the latest housekept version of the tree, not the highest-ranked item from the current tree.

```
cycle 1: push pkt 1 (lo)
cycle 2: push pkt 2 (lo)
...
cycle 10: push pkt 10 (lo)
...
cycle 95: housekept "push pkt 1"
...
cycle 99: housekept "push pkt 2"
...
cycle 984: housekept "push pkt 10"
cycle 10,000: push pkt 11 (hi)
cycle 10,001: pop
...
cycle 10,050: housekept "pop"
cycle 11,093: housekept "push pkt 11"
...
```

And now it's very clear that the `pop` on cycle 10,001 should release `pkt 1`.

anshumanmohan, 2023-09-18:
In today's synchronous meeting we firmed up the plan a little more when it comes to:
- What components we'll have.
- How they'll be orchestrated.

I'm writing this out below.

1. _The PIFO tree._ This component is invoked dynamically by the _main loop component_ (see below) via the use of `ref` cells and `invoke`s. In turn, this component may statically invoke the _stats component_ (see below) to tell it some new piece of information gleaned from its own invocation. For a running example, let's say it reports nothing for pop and peek, and for push it reports how it classified the packet (red/blue).
2. _The main loop._ Unchanged from before. It dynamically invokes the PIFO tree component and stashes any answers it gets in return. It interacts with no other components.
3. _The stats component._ This is new. It has some simple adders and registers that compute and store some "accumulator"-ey statistic. It is invoked by the PIFO Tree component as described above, and by the mother as described below. It does not interact with the main loop. For our running example, let's say it stores simple counts of red/blue.
4. _The mother_. This is new. It periodically, and statically, calls the statistics component and grabs its data from it. In the future, it may then use this information to say something to the PIFO tree component, but that's not part of our running example or MVP. That is, for now it only interacts with the stats component and no other.

PR #1736 (2023-11-16): PIFO Trees: Telemetry
This is WIP and is not ready for serious review. I am just opening this PR so I can seek help getting it all the way there. This PR will eventually close #1697: the overall plan is exactly as summarized at the [bottom](https://github.com/cucapra/calyx/issues/1697#issuecomment-1724403886) of that thread.

Important background:
- Recall that what we call "PIFO" in [`pifo.py`](https://github.com/cucapra/calyx/blob/52c56353fe1bd4bcd419d15e99ea61b2e3b40e89/calyx-py/test/correctness/pifo.py#L47) in Calyx is actually a more general thing: it is simply a queue that orchestrates t [... cut]
closed 2023-11-16 by anshumanmohan