calyxir/calyx #1714: Calyx docker image does not have git initialized in /home/calyx

nathanielnrn, 2023-09-04:
In trying to get the one line of code change in #1698 to work it seems like the docker image ghcr.io/cucapra/calyx:0.6.0 does not have a git repo initialized at /home/calyx.

This can be verified on havarti which has the 0.6.0 docker image.

sampsyo, 2023-09-08:
Hmm… I wonder if a different (although also possibly annoying) approach might fix this sort of problem. Namely, our current approach to GitHub Actions is to rely on a specific, tagged version of the underlying Docker container:
https://github.com/cucapra/calyx/blob/22a2e6b5fe10ad8e1daa9bf1ca9769f048cd54d5/.github/workflows/rust.yml#L15

That is fundamentally a little weird because it means that new changes on a given branch (including `master`, but also feature branches) aren't reflected in that container. It also means that we have to manually bump this version every time we want the basic environment to be updated. Finally, there are some potential weirdnesses that arise from having two different versions of Calyx built: the one in the Docker container and the one we `git clone` and build.

It seems to me like there are two logical resolutions to this:

1. Use a "fresh" Docker image to run the tests. That would mean that the `build` workflow depends on the `depot` workflow: we first build a new Docker image with all the latest code, and then we run the tests in a container based on that image.
2. Stop using the "full" (release) Docker image for CI altogether. We would (probably) still run the CI stuff in a Docker container, but that would be based on a "dev container" that just has the _dependencies_ we need—not a pre-built Calyx installation that we then immediately throw away. This could even use a Dockerfile that is shared with the "full" Docker image build if we want to avoid duplicating work.

It seems to me like these might make a tad more sense that making our released Docker image contain a git checkout?

rachitnigam, 2023-09-09:
Yeah, this is a tricky one. Maybe another possibility is having some mechanism to detect/specify that a particular PR changes the Dockerfile and therefore requires a complete rebuild before running the tests. The goal would be to have a fast path for most PRs that don't touch the Dockerfile.

rachitnigam, 2023-12-24:
This might be a problem with depot.dev itself. Seems like it is not adding the `.git` file when building the image

rachitnigam, 2023-12-24:
> It seems to me like these might make a tad more sense that making our released Docker image contain a git checkout?

@sampsyo The problem is that `calyx` also contains dependencies like `fud` which we do want to be installed. If we stop putting the Calyx repo in the container, the workflow files have to do the job of setting up the tools which might leave us in the same position as before: the test environment for Calyx would be different from the containers we release.