calyxir/calyx #1721: xclrun.py does not correctly handle fixedpoint programs.

nathanielnrn, 2023-09-09:
In exploring what kinds of programs work/don't work well on fpgas as part of going through the calyx -> FPGA flow we currently have, it was found that [xclrun.py](https://github.com/cucapra/calyx/blob/master/fud/fud/xclrun.py) does not currently handle fixedpoint inputs correctly.

Specifically, trying to run the calyx output of the [systolic array tests](https://github.com/cucapra/calyx/tree/master/tests/correctness/systolic/output) fails to execute due to trying to coerce an array into a scalar. This was an easy fix, but even with that there seem to be more issues. And the output after execution looks like this:

```
  "memories": {
    "l0": [
      null,
      null,
      null
    ],
    "l1": [
      null,
      null,
      null
    ],
  //... more memories
```

I suspect that this may be due to a a type conversion mismatch happening [here](https://github.com/cucapra/calyx/blob/master/fud/fud/xclrun.py#L42-L47).

rachitnigam, 2023-09-10:
Oh, that’s funky! Maybe it would be useful to see what the raw dumped data looks like. The right thing to do would be to use the same infrastructure that the simulation stuff uses (#1715).

sampsyo, 2023-09-10:
Argh; that is annoying! And probably my fault (as the person who introduced `xclrun` to replace the funky built-in fud stuff).

One upside to `xclrun` being independent is that, hopefully, this should make the debugging loop a little faster. Namely, it should be possible to do this:

* Use fud to produce an `xclbin` from a given benchmark.
* Directly invoke `xclrun` on that `xclbin`, not using fud, to try running the simulation.
* Make a change to `xclrun` (e.g., to dump out the raw input/output data, as @rachitnigam suggests) to see what happens and repeat.

Happy to help with this!