calyxir/calyx #1724: `fud` Fails to Run `xclbin`s Targeted for Hardware on FPGAs

nathanielnrn, 2023-09-17:
Seems like I was [wrong](https://github.com/cucapra/calyx/discussions/1701#discussioncomment-7024372) in saying that fpga execution was working. Some configurations I _thought_ were making my xclbin compilations target hardware were not doing that, but rather still targeting hardware emulation. Sorry about the mixed messages.

PYNQ is failing to find [`active_devices`](https://github.com/Xilinx/PYNQ/blob/master/pynq/bitstream.py#L88) and is [raising a RuntimeError as a result](https://github.com/Xilinx/PYNQ/blob/master/pynq/pl_server/device.py#L71). Spent some time digging at this but unfortunately some [suggestions](https://discuss.pynq.io/t/error-no-device-found-when-loading-overlay/2989/7) I found don't seem to solve things. Some things went a bit [over my head](https://discuss.pynq.io/t/pynq-3-0-1-can-not-loading-overlay-no-device-found/4955/4) but I will dig more into them and see what I can come up with.

cc @sampsyo as the last person to work on this AFAIK.

nathanielnrn, 2023-09-19:
For sake of completeness, sharing some steps to repo and some error logs. There is a new error compared to initial issue. Now Havarti is "[failing to lock device](https://github.com/Xilinx/PYNQ/blob/master/pynq/pl_server/xrt_device.py#L520-L522)."

An xclbin was created targeting hardware. This required changing part of the fud configuration to

```
[stages.xclbin]
mode = "hw"
device = "xilinx_u50_gen3x16_xdma_201920_3"
temp_location = "/tmp"
ssh_host = ""
ssh_username = ""
```
(Everything included for completeness, relevant line is `mode = "hw"`.)

Afterwords, invoking the following:
`fud e dot-product-hw-mode.xclbin --to fpga --from xclbin -s fpga.data <path>/examples/dahlia/dot-product.fuse.data -vvv`

Leads to the following stacktrace

```
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/scratch/nrn25/data/calyx_fpga/lib/python3.8/site-packages/fud/xclrun.py", line 144, in <module>
    xclrun()
  File "/scratch/nrn25/data/calyx_fpga/lib/python3.8/site-packages/fud/xclrun.py", line 136, in xclrun
    out_data = run(Path(args.bin), in_data)
  File "/scratch/nrn25/data/calyx_fpga/lib/python3.8/site-packages/fud/xclrun.py", line 81, in run
    ol = pynq.Overlay(str(xclbin.resolve(strict=True)))
  File "/scratch/nrn25/data/calyx_fpga/lib/python3.8/site-packages/pynq/overlay.py", line 348, in __init__
    self.download()
  File "/scratch/nrn25/data/calyx_fpga/lib/python3.8/site-packages/pynq/overlay.py", line 428, in download
    super().download(self.parser)
  File "/scratch/nrn25/data/calyx_fpga/lib/python3.8/site-packages/pynq/bitstream.py", line 168, in download
    self.device.download(self, parser)
  File "/scratch/nrn25/data/calyx_fpga/lib/python3.8/site-packages/pynq/pl_server/xrt_device.py", line 541, in download
    self._xrt_download(data)
  File "/scratch/nrn25/data/calyx_fpga/lib/python3.8/site-packages/pynq/pl_server/xrt_device.py", line 522, in _xrt_download
    raise RuntimeError("Could not lock device for programming - " + str(err))
RuntimeError: Could not lock device for programming - -19
```

Tried looking for error code -19 but a brief google search did not find much w.r.t xrt error codes.

sampsyo, 2023-09-20:
Cool; this really could be worse! It kinda feels like it could be a permissions issue or something.

Anyway, I might recommend trying to debug this outside of fud. That means executing xclrun manually, doing something like this:

```sh
$ source /scratch/opt/Xilinx/Vitis/2020.2/settings64.sh
$ source /opt/xilinx/xrt/setup.sh
$ python3.8 fud/xclrun.py <something> foo.xclbin
```

(You can probably get the actual `xclrun` command, in the last line, from a verbose fud execution.) One random thing to try would be just doing that under a `sudo` to see if it wants you to be the root user to access the real FPGA. Another could be to try confirming that XRT can see the FPGA device at all… for example, XRT's `xbtop` command is supposed to just list the attached cards:
https://xilinx.github.io/XRT/2023.1/html/xbtop.html

Maybe poking around at some of those commands would reveal if there's something obviously wrong with the whole FPGA setup in havarti?

sampsyo, 2023-09-24:
I ended up with a moment to look into this! I reproduced stuff as described above, and I first ran into the same issue about the `XCL_EMULATION_MODE` environment variable as you, @nathanielnrn. I embarrassingly did not notice that you had a fix [on a branch](https://github.com/cucapra/calyx/compare/master...fix-pynq) and ended up recreating the same idea myself, with some extra bells and whistles, in #1728. If that looks right to you, @nathanielnrn, please merge at will.

Then I went about reproducing your "Could not lock device for programming" trouble. The TL;DR is that you can work around this problem on havarti by setting this in your fud config:

```yaml
[stages.fpga]                                                                                      
xrt_location = "/opt/xilinx/xrt" 
```

Details below.

---

The key was indeed running `xclrun` "by itself," which amounted to this command:

```sh
XRT_INI_PATH=`pwd`/xrt.ini bash -c 'source /scratch/opt/Xilinx/Vitis/2020.2/settings64.sh ; source /scratch/opt/xilinx/xrt/setup.sh ; python3.9 -m fud.xclrun lti.hw.xclbin examples/tutorial/data.json'
```

That revealed a new, more specific error before the Python program crashed:

```
[XRT] ERROR: Failed to open /sys/bus/pci/devices/0000:3b:00.1//kds_numcdmas for reading: No such file or directory
```

Clearly, something is going wrong when the XRT user-space library tries to communicate with the PCI card (via its [associated kernel driver](https://xilinx.github.io/XRT/master/html/sysfs.html)). I investigated a little bit and found that a file there called `kds_numcdma` exists, but not the plural. Curious!

A little googling lead me to https://github.com/Xilinx/XRT/pull/6056, the XRT PR where this changed on the user-space side. (Thank goodness this library is open source; I might otherwise never have figured it out.) The clear conclusion is that we are dealing with a version mismatch: the kernel module is old (using `kds_numcdma`) and the user-space library is new (using `kds_numcdmas`).

While I don't exactly know why this happens, there is a conspicuous underlying reason. Havarti has two versions of the XRT library installed on it, at `/opt/xilinx/xrt` and `/scratch/opt/xilinx/xrt`. The former is old (2.8.743, aka 2020.2) and the latter is newer (2.13.0, built 7/28/2022). I guess the former must be the one whose kernel modules are currently active.

A _good_ way to fix this would be to just install a fresh 2023 version of XRT and make sure that all its pieces are up to date, and maybe delete the old versions if we're sure we don't need them. (If either @nathanielnrn or @evanmwilliams wants to take this thankless task up, that would be awesome.) But a lazy solution is just to switch to using the old version of the XRT userspace library, by changing your fud config as depicted above.

closed 2023-09-24 by nathanielnrn

nathanielnrn, 2023-09-24:
This is great! Thank you for looking into this. I spent much of yesterday looking into a `EINVAL (22) Invalid argument` error but this seems to have short circuited the need to look into that. I will try to look into fresh-installs of xrt but that may be a few weeks out at this point.

PR #1728 (2023-09-24): [fud] Fix environment variables for real-FPGA runs
This should probably fix #1724, superseding the `fix-pynq` branch (apparently @nathanielnrn and I stumbled on the same issue). We were setting the environment variables like `XCL_EMULATION_MODE=hw` in hardware-execution mode, where what we should be doing is not setting this at all in this mode.

To confirm that this works (and still works for emulation too), I produced two `xclbin`s from the same Calyx program: one for emulation (`lti.xclbin`) and one for real hardware (`lti.hw.xclbin`). Then I ran the two necessary fud commands to execute th [... cut]
commit ecb79009 (2023-09-24): [fud] Fix environment variables for real-FPGA runs (#1728)

This should probably fix #1724, superseding the `fix-pynq` branch
(apparently @nathanielnrn and I stumbled on the same issue). We were
setting the environment variables like `XCL_EMULATION_MODE=hw` in
hardware-execution mode, where what we should be doing is not setting
this at all in this mode.