calyxir/calyx #1725: Determining the static interface for `calyx` components

paili0628, 2023-09-18:
We should make a decision on exactly what the interface for the `static component` would look like. In particular, there are 3 options:
1. have a `go` signal that triggers for exactly 1 cycle. The component will be 'ready' to accept inputs again after n cycles (i.e., the static latency refers to the II). (No done signal)
2. have neither a `go` nor a `done` interface. The output ports hold the outputs relevant to the value on the input ports exactly n cycles ago. In this case the component will have to be 'ready' to accept inputs every cycle, since there's no way to distinguish between valid and invalid signals.
3. have a `go` signal that has to be continuously triggered until a `done` signal is up (the static latency refers to how many cycles have to elapse since the first cycle the `go` signal is triggered until the `done` signal is up).

Correspondingly, there are also two options to define the dynamic interface:
1. The classic calyx `go-done` interface with no pipelining.
2. a `go-ready-done` interface, where the `go` still has the same semantic meaning, the `done` means that a valid output is produced on the output ports, and the `ready` means that the component is now 'ready' to accept inputs (i.e., the component only 'accepts' inputs when `go==1'd1 & ready==1'd1`.

calebmkim, 2023-09-20:
Thanks for writing this up, one small question on 2.: 

> 2. have neither a go nor a done interface. The output ports hold the outputs relevant to the value on the input ports exactly n cycles ago. In this case the component will have to be 'ready' to accept inputs every cycle, since there's no way to distinguish between valid and invalid signals.

Is it necessary for the component to be ready to accept inputs every cycle? In other words, can't we just make it the Calyx programmer's responsibility to use the component correctly if the component's II is not 1? The only reason why I say this is that it seems like we wouldn't be able to represent a component that has an II of 2 or more in this case. Lmk if there's something I'm missing. 

**Another question:** 
Is there another option that is kind of a mix of 1. and 3.? More concretely,  the `n` in `static<n> component_name` in this option represents the latency of the component as in 3., but there is also no done signal for the component like in 1. This is just my opinion so feel free to disagree, but in some ways I feel it's the most intuitive option since it matches Calyx groups most closely (e.g., `static <n> group_name` means that the group's latency is `n` cycles, and it does not have a done signal).

rachitnigam, 2023-09-20:
I think (2) is not tenable because if your module has II > 1, then you cannot correctly use it because you have no way to know when it started its first execution.

sampsyo, 2023-09-27:
Excellent summary. I think that's exactly where we stand. I guess the strawman plan here could be to just live with the asymmetry between components and groups... do folks know of bad problems with that? Maybe inlining?

calebmkim, 2023-10-03:
### Decision 
Just going to update this with our synchronous discussion: we decided that the asymmetry between groups and components is (probably) acceptable, since there is an asymmetry in between how components and groups are used: components can be pipelined in their execution, while groups cannot. Therefore, we can (at least for now) try to implement the interface and we'll live with the asymmetry.  

### Question 
One question that I just thought of: currently, we infer the latency of registers and multipliers by looking at the `@static` annotation on their `go` port, even though they have a `done` port. This `@static` annotation is just a way of saying "we know the output will be available after one cycle"; kind of like a hint for the compiler. Couple questions on this: 

1. Should we keep this behavior, in light of our decision about the static interface components? If we do, there will be a distinction between two things: 1) the actual `static<n>` interface (which is to trigger `go` for one cycle, wait `n` cycles, no `done` signal) and 2) the `@static(n)` annotation, which is just a hint to promote to static. 
2. What should we do about dynamic components whose entire control program we can infer the latency of? We probably won't be able to upgrade the component to a `static<n> component` (since that would involve getting rid of the `done` signal), but we may be able to attach the `@static(n)` annotation "hint" to its ``go'' port. This will help in the `static-promotion` of groups.

rachitnigam, 2023-10-03:
Thanks for following up on this @calebmkim! Thoughts on "definitely static" primitives like registers:
1. On the elegance front, I think it would be nice to say something like: all registers are completely static, a fact which is reflected in their primitive because use the `static` keyword. They do not have the `done` signal and the only way to use them is in the context of `static` groups. The Calyx compiler knows how to instantiate the `done` signal for such groups and will do the right thing.
2. On the pragmatic front: we can keep the `done` signal for the register and say, "well actually, because it is so common to use registers in either context, we have implemented a *really efficient* `done` signal for the register and the compiler is allowed to assume that the `done` interface works correctly and can be eliminated in static contexts".

My point in mentioning (1) & (2) is that, notionally, we should think of `static` primitives as truly static: they do not have need or have a `done` signal. The fact that they have a `done` signal is *merely* an optimization that's available for the compiler to leverage. It should be totally okay to *ignore* the done signal and instantiate a `static group` FSM to expose the `done` signal as well because first and foremost, `std_reg` is a static primitive.

Does this philosophy resonate?

rachitnigam, 2023-10-03:
@andrewb1999 agrees that the new "assert `go` for one cycle" interface is a good idea and so we should commit to implementing it. Once it lands, Andrew will update AMC to use the new interface and hopefully things will continue working.

sampsyo, 2023-10-04:
> My point in mentioning (1) & (2) is that, notionally, we should think of `static` primitives as truly static: they do not have need or have a `done` signal.

Right, this seems like the way to me! To put it another way, we want to have two nonoverlapping categories of components:

1. Dynamic components, which can only be used in a dynamic context (and have a `done` port).
2. Static components, which can only be used in a static context (and do not have a `done` port). (But you can wrap them to use them in a dynamic world!)

Then we might want to create a small third category for special cases:

3. Stuff like `std_reg` that can be used in _either context_, without wrapping. These have _both_ a known static latency _and_ a `done` port. When used in a static context, you just ignore the `done` port completely.

One could argue that we should eliminate category 3, and create two different primitives called `static_reg` and `dyn_reg` for example. But it doesn't seem too bad to have a small number of things in category 3, as long as we don't make that the common case!

rachitnigam, 2023-10-04:
Right! The argument for keeping category 3 is that name punning for very common things like registers is useful and we expect synthesis tools to remove the unconnected `done` register when used in static context anyways.

calebmkim, 2023-10-04:
One more note: for `static-promotion`, it seems like we would be promoting some components from (1) to (3). Is this fine? I know we should probably keep (3) small; on the other hand, this could help with static promotion.

rachitnigam, 2023-10-05:
Hm, not sure I follow. If you promote something, it goes from 3 to 1 right? This is because we're saying that we definitely don't need to use the `done` signal

paili0628, 2023-10-05:
I think we have to rethink what the latency of a static component actually means, especially for pipelined components. 

Writing a pipelined component would probably have a control program that looks something like this: 
```
static par {
  static seq {// 1st stage of pipeline } 
  static seq {// 2nd stage of pipeline } 
  ... // etc. 
} 
```
But in this case, the latency of the control program would be the _II_, not the _latency_. 

The reason we cannot represent a pipeline using a `static seq` is that we only have one fsm, which can only be at one value, which means we cannot represent multiple stages.

rachitnigam, 2023-10-06:
@paili0628 My understanding of the annotation is exactly the same! That is, for sequential components, the annotation represents both the II and the latency but for pipelined component, it tells you when you can re-invoke the module. 

@calebmkim thoughts?

calebmkim, 2023-10-06:
Yeah I agree with the above^. I think this all sounds good.

calebmkim, 2023-10-06:
Yeah, that's exactly what I was thinking.

rachitnigam, 2023-10-14:
Should we add a new `@latency(n)` attribute to indicate the latency of a particular port? I was thinking something like this could be checked in the future and be useful to ensure that the pipeline is correctly balanced.

sampsyo, 2023-10-16:
It's a pretty good idea. This would end up being, like, a super restrictive embedding of a subset of Filament types… as in, a Filament component with a simple signature something like this:

```
comp mycomp<'G: X>(
  go: interface['G],
  inport: ['G, 'G+X] 32,
) -> (
  outport: ['G+Y, 'G+Y+1] 32,
)
```

Could compile to a Calyx signature like this:

```
static<X> component mycomp(
  inport: 32
) -> (
  @latency(Y) outport: 32
)
```

rachitnigam, 2023-12-12:
Subsumed by #1754

closed 2023-12-12 by rachitnigam