calyxir/calyx #1726: Optimizing programs in the presence of an observable global clock

hackedy, 2023-09-19:
I think we've talked about this before but I wanted to commit a concrete example to the repo so we can have something to point at when we talk about it.

I expect the equation
```
seq { g1; g2 } = par { g1; g2 }
```
should hold when `g1` and `g2` use disjoint resources. Rewriting occurrences of the left side of the equation to the right side of the equation would reduce the time the program takes to run, so it looks like a nice optimization.

However, there are program contexts in which these programs are distinguishable because you can (so to speak) stand next to the program with a stopwatch and see how long it took to run.

There's a program [timing-observations.futil](https://github.com/cucapra/calyx/blob/vcalyx/vcalyx/examples/timing-observations.futil) that illustrates this. The seq_writes and par_writes components are exactly the same, except their control blocks use seq and par blocks respectively. The main component times them with a "stopwatch" gadget and records the elapsed time in external memories.

They take observably different amounts of time, so the program writes two different values to the elapsed time memories.
```
% fud exec vcalyx/examples/timing-observations.futil --from calyx --to interpreter-out -s interpreter.flags " -q" -s verilog.data vcalyx/examples/timing-observations.futil.data -q | jq .memories
{
  "main": {
    "seq_mem": [
      4
    ],
    "par_mem": [
      1
    ]
  }
}
```

So `seq { g1; g2 }` is not doing the same thing as `par { g1; g2 }` here even though the two groups use disjoint resources. So it seems like a reasonable optimization is violating contextual equivalence.

A few questions to think about. I think 1. is what we've suggested doing before, if I remember correctly. I think the answer to 2. is "no" and the answer to 3. is "probably not" but answering 1. will require actually writing down some math and seeing that it genuinely fixes the problem. Also there's a question 0. "is this example program doing what I think it's doing?"

1. Should the semantics of Calyx be nondeterministic, and is this apparent disagreement the result of resolving that nondeterminism in two different ways?
2. Are the two programs actually not contextually equivalent?
3. Are there static analyses or type systems that rule out the "stopwatch" gadget and things like it? Do we care?

rachitnigam, 2023-09-19:
Thanks for starting the discussion on this! The short answer is that the dynamic fragment of Calyx is explicitly allowed to change the timing behavior of your program and only guarantees that order of operations will be preserved. The new `static` fragment of Calyx is the one that should be used when implementing programs that are time-sensitive. Answering your question in order:
1. In presence of data races, the semantics of Calyx are undefined. In dynamic calyx, these may occur any time threads in a `par` block use the same resources. The compiler is free to reorder their execution however it feels and therefore will cause observable changes. If there are no data races, dynamic Calyx only provides guarantees on ordering of things.
2. They *are* but we have to be precise about the contexts that are allowed. Contexts that can use time to distinguish programs are not allowed.
3. Tough to say and we probably don't care. With the addition of the `static` fragment, users have the ability to opt into time-sensitive behavior. Even then, the design does not allow access to a global clock; only a local one which is important for scalable optimization.

You might be interested in looking at #1722 as an example of an optimization enabled by the `static` fragment and maybe ping the Calyx channel if you want to read a draft of the static paper.

hackedy, 2023-09-20:
Are there data races in the example programs? I've split it into the "optimized" [par version](https://github.com/cucapra/calyx/blob/vcalyx/vcalyx/examples/timing-par.futil) and the "unoptimized" [seq version](https://github.com/cucapra/calyx/blob/vcalyx/vcalyx/examples/timing-seq.futil).

hackedy, 2023-09-20:
Contextually equivalent means "equivalent in all well typed contexts"--if we want to avoid the use of time in distinguishing contexts we have to do it with a type system (or static analysis).

rachitnigam, 2023-09-22:
I don't think there are any data races in the programs. I just wanted to mention that facet as another thing to consider. The interpreter implements one semantics that is a refinement of the DRF0 semantics I mentioned but the compiler implements a different one so you might want to play with verilog simuation.

> Contextually equivalent means "equivalent in all well typed contexts

I guess that is a very type directed definition of context equivalence. I normally think of context equivalence along the lines of what kinds of behaviors can the context observe (non-termination, memory). In this case, contexts that can observe timing can obviously distinguish between the programs because the Calyx semantics allows the compiler to change the time semantics of the program.

hackedy, 2023-09-27:
I think maybe the solution here is to give programs a nondeterministic semantics that captures behavior "up to arbitrary delays." So the clock gadget in my example would not have a single legal trace where the outputs on `out` are
```
1, 2, 3, 4, 5,
```
but instead would be a set of traces 
```
1, 2, 3, 4, 5,
1, 1, 1, 1, 2, 2, 3, 4, 5, ...
```
where the n-fold repetitions of numbers are happening because the register is taking n cycles to produce a `done` signal.

Then the observed different behaviors are just two behaviors that sit in the same set. I hope!

rachitnigam, 2023-09-28:
This certainly jives with latency-insensitive formalism that uses a notion of "stallable" modules to formalize valid compositions of modules

rachitnigam, 2023-10-03:
@hackedy if this issue feels answered, let's close it

closed 2023-11-11 by hackedy