calyxir/calyx #1737: Re-enable playground github workflow

rachitnigam, 2023-10-11:
The playground workflow [suddenly started failing](https://github.com/cucapra/calyx/actions/runs/6463043645/job/17546014541). A likely cause is the underlying ubuntu container for GH actions being updated and causing one of the dependencies to fail.

sampsyo, 2024-02-27:
The Hip New Way to do this kind of thing is to use the WebAssembly component model and [jco][]! If anyone's interested, overhauling the build pipeline to use that could be a good way to go. (Another option is to just fix the current build pipeline in its current form.)

[jco]: https://github.com/bytecodealliance/jco

YJDoc2, 2024-03-04:
Hey, after merge of https://github.com/calyxir/calyx/pull/1949 , the playground deployment is again running, but has some issues like :
1. Some examples are not found : eg Minimize regs simple , resource sharing etc . 
2. some passes are not valid : eg component-interface-inserter , resource-sharing

The simple fix would be to check which files for examples are missing / moved and update the config.json file accordingly ; and remove the invalid passes. For a more long-term fix, I'm thinking of a simple node script which will basically do the above automatically, and can be run in CI on each push to validate that the PR does not break anything.

I also thought to maybe generate the config.json file automatically, i.e. a script which will scan the tests dir to find the examples present and generate the file from that, and can be run in the website build command to always have up-to-date examples and passes. But,
1. There are a lot of tests, and I think adding all of those as example on site would overwhelm the users (especially first time users - like me :p ) 
2. I don't have much idea how the default passes can be determined from the file

So, I feel only the validation CI would be good.

Apart from that is there anything this issue needs to address ?

Thanks :)

sampsyo, 2024-03-09:
That validation thing seems like a pretty good idea! It would hopefully be pretty straightforward to check that the list of passes actually exists. And you're right that including all the tests as examples in the playground probably wouldn't make sense—it seems more reasonable to keep these "handpicked," and checking that they actually exist would be lovely.

For the set of passes, it would be possible to parse a list of passes from the output of `calyx pass-help`. This output is pretty long, however, so I'm not 100% sure how convenient it is to parse? For a default set of passes to surface in the UI, it might make sense to include the `compile` alias that this output mentions:

```
- compile: static-inline, merge-assigns, dead-group-removal, simplify-static-guards, add-guard, compile-static-interface, dead-group-removal, compile-static, tdcc
```

PR #1976 (2024-03-21): Add playground config validation script and CI
closes #1737 
ref : https://github.com/calyxir/calyx/issues/1737#issuecomment-1986905609

Sorry it took a bit of time to get this up, got busy in other things...

This adds a node script which will validate that 
1. All passes listed in the passes.json are valid (cross checking with calyx pass-help o/p)
2. all example files that are in config.json are actually present

Currently parked the CI in rust.yaml, as I figured it would be better to run on each push, can move to other yaml as per suggestions.

Also instead of checking with master, the file [... cut]
closed 2024-03-21 by rachitnigam