calyxir/calyx #1738: Unexpected calyx.std_extsi in fixed-point multiplication

zzy666666zzy, 2023-10-12:
Hi there,

I am trying to lower a MLIR that implement a 32-b fixed point convolution algorithm. The lowering path I am using is 
```
mlir-opt --lower-affine
mlir-opt --scf-for-to-while
circt-opt --lower-scf-to-calyx --canonicalize --verify-diagnostics
circt-translate -export-calyx --verify-diagnostics
```
error occurred at the last stage. I have checked this [issue ticket](https://github.com/llvm/circt/pull/3189) but I am not quite sure how to map `std_extsi` to `std_pad`. Is there any suggestions about this? Thanks a lot!

Below is the input MLIR 

```
  func.func @mlir_func_ZTSZZ4mainENKUlRN4sycl3_V17handlerEE_clES2_E7conv_2D(%arg0: memref<841xi32, 1>, %arg1: memref<49xi32, 1>, %arg2: memref<144xi32, 1>) {
    %c65536_i64 = arith.constant 65536 : i64
    %c2_i64 = arith.constant 2 : i64
    %c32768_i64 = arith.constant 32768 : i64
    %c0_i32 = arith.constant 0 : i32
    affine.for %arg3 = 0 to 12 {
      affine.for %arg4 = 0 to 12 {
        %0 = affine.for %arg5 = 0 to 7 iter_args(%arg6 = %c0_i32) -> (i32) {
          %1 = affine.for %arg7 = 0 to 7 iter_args(%arg8 = %arg6) -> (i32) {
            %2 = affine.load %arg1[%arg7 + %arg5 * 7] : memref<49xi32, 1>
            %3 = arith.extsi %2 : i32 to i64
            %4 = affine.load %arg0[(%arg5 + %arg3 * 2) * 29 + %arg4 * 2 + %arg7] : memref<841xi32, 1>
            %5 = arith.extsi %4 : i32 to i64
            %6 = arith.muli %3, %5 : i64
            %7 = arith.divsi %6, %c32768_i64 : i64
            %8 = arith.remsi %7, %c2_i64 : i64
            %9 = arith.divsi %6, %c65536_i64 : i64
            %10 = arith.addi %8, %9 : i64
            %11 = arith.trunci %10 : i64 to i32
            %12 = arith.addi %arg8, %11 : i32
            affine.yield %12 : i32
          }
          affine.yield %1 : i32
        }
        affine.store %0, %arg2[%arg4 + %arg3 * 12] : memref<144xi32, 1>
      }
    }
    return
  }
```

rachitnigam, 2023-10-12:
Thanks for opening the issue! This is because Calyx is currently missing a sign extension primitive (https://github.com/cucapra/calyx/issues/1009). I've opened https://github.com/cucapra/calyx/pull/1739 to add this primitive and will update the CIRCT repo to add the lowering needed next.

rachitnigam, 2023-10-12:
Okay, I've added the required changes to the CIRCT repo: https://github.com/llvm/circt/pull/6285. Once that PR is merged, the above program is correctly lowered using the new `std_signext` primitive that I just added to Calyx. You'll need the latest version of both Calyx and CIRCT once that PR is merged to generate the program.

rachitnigam, 2023-10-12:
Okay, the two PRs are merged and the program should lower correctly. I'm going to close this issue now but please try it and feel free to reopen the issue if you run into any problems.

zzy666666zzy, 2023-10-13:
Thanks a lot for your reply! 
`arith.extsi` can be lowered using `circt-translate -export-calyx --verify-diagnostics` with `std_signext`.

Because I am intending to generate the Verilog code, I used following lowering commands afterwards
```
calyx <input> -p validate -p discover-external -p validate -x discover-external:default=$1 -x discover-external:strip-suffix=_ -o <input> 

calyx -l $Calyx_PATH -b verilog --synthesis --disable-verify --nested <input> -o <output>

calyx -l $Calyx_PATH -b xilinx --synthesis --disable-verify --nested <input> -o <output>

calyx -l $Calyx_PATH -b xilinx-xml --synthesis --disable-verify --nested <input> -o <output>
```
I got the error
```
23 |    std_signext_1 = std_signext(32, 64);
   |    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ Undefined primitive or component name: std_signext
```
while executing `calyx <input> -p validate -p discover-external -p validate -x discover-external:default=$1 -x discover-external:strip-suffix=_ `. I `grep std_signext` in calyx repo, it seems `std_signext`is only in hardware primitives (.sv), not implemented in `*.rs`. Maybe I am wrong, it was implemented somewhere else. So is there any suggestions on generating the Verilog?

Many thanks!

rachitnigam, 2023-10-13:
Can you make sure you've pulled the latest version of the Calyx repository?

zzy666666zzy, 2023-10-13:
Thanks @rachitnigam . 

Yes, the version I pulled is up-to-date. `std_signext` is in `binary_operators.sv` and `binary_operators.futil`. But the error still exists `Undefined primitive or component name: std_signext.` I tried to `grep` similar primitives like `std_pad` and `std_mux`, they are in `memory-with-external-attribute.expect` and `./tests/import/a.expect`, but `std_signext` is not. I tried to manually add them in these files but still no luck.

Or it is because of the termination by error message, can we turn it off somewhere?

rachitnigam, 2023-10-13:
Hm, my guess is that it's not using the right location for the library. Adding `std_signext` to those files won't do anything; they are merely tests.

Could you do the following:
1. Run the `git rev-parse HEAD` to show the commit the repository is on
2. Run the calyx command with `--log info` and paste the output from it? It should show where the `primitives/binary_operators.futil` is being picked up. If it is not the same location pointing to the `binary_operators.futil` that contain `std_signext` then that's our problem.

zzy666666zzy, 2023-10-13:
Thanks @rachitnigam 

1. `git rev-parse HEAD` returns `31c3cc5d14f4991b8284b8bda7b415254433301c`.
2. `--log info` helps, it points to my current folder instead of `Calyx` folder. Now `std_signext` seems can be lowered, but a new error came out `thread 'main' panicked at 'attempt to shift left with overflow', calyx-ir/src/builder.rs:165:19`.

rachitnigam, 2023-10-13:
Yeah, that error was just fixed in the latest commit. Can you pull one more time and see if that gets rid of the error?

zzy666666zzy, 2023-10-16:
Thank you so much @rachitnigam 

The MLIR is correctly lowered and outputs the correct result on the board. Just a quick follow-up question.

Given the memory depth of the function's argument 
%arg0: memref<**841**xi32, 1>, %arg1: memref<**49**xi32, 1>, %arg2: memref<**144**xi32, 1>,
based on my understanding we could only make all IO pointers' depth the same constant by using 
`calyx <input> -p validate -p discover-external -p validate -x discover-external:default=DEPTH -x discover-external:strip-suffix=_ -o <input> `, I make `DEPTH` to 1024 to cover three pointers' depth. But it will generate redundant hardware.

Is there any chance that we could assign the exact depth to each IO pointer to generate the IO BRAM with the exact depth?

Many thanks!

rachitnigam, 2023-10-17:
Awesome! Yeah, this is unfortunately messy and right way to fix it is to support `ref` definitions in CIRCT (https://github.com/llvm/circt/issues/4831). If this becomes a blocking problem for your use case, please open an issue and I can try to provide some sort of workaround in the meantime.

closed 2023-10-17 by rachitnigam