calyxir/calyx #1747: Clarify exact IO depth when lowering MLIR

zzy666666zzy, 2023-10-18:
Awesome! Yeah, this is unfortunately messy and right way to fix it is to support `ref` definitions in CIRCT (https://github.com/llvm/circt/issues/4831). If this becomes a blocking problem for your use case, please open an issue and I can try to provide some sort of workaround in the meantime.

_Originally posted by @rachitnigam in https://github.com/cucapra/calyx/issues/1738#issuecomment-1767092853_

zzy666666zzy, 2023-10-18:
Hi @rachitnigam 

Just opened a new issue following this [issue](https://github.com/cucapra/calyx/issues/1738#issuecomment-1764732713)

The original MLIR with different arguments' depth:
```
  func.func @main(%arg0: memref<841xi32, 1>, %arg1: memref<49xi32, 1>, %arg2: memref<144xi32, 1>) {
    %c65536_i64 = arith.constant 65536 : i64
    %c2_i64 = arith.constant 2 : i64
    %c32768_i64 = arith.constant 32768 : i64
    %c0_i32 = arith.constant 0 : i32
    affine.for %arg3 = 0 to 12 {
      affine.for %arg4 = 0 to 12 {
        %0 = affine.for %arg5 = 0 to 7 iter_args(%arg6 = %c0_i32) -> (i32) {
          %1 = affine.for %arg7 = 0 to 7 iter_args(%arg8 = %arg6) -> (i32) {
            %2 = affine.load %arg1[%arg7 + %arg5 * 7] : memref<49xi32, 1>
            %3 = arith.extsi %2 : i32 to i64
            %4 = affine.load %arg0[(%arg5 + %arg3 * 2) * 29 + %arg4 * 2 + %arg7] : memref<841xi32, 1>
            %5 = arith.extsi %4 : i32 to i64
            %6 = arith.muli %3, %5 : i64
            %7 = arith.divsi %6, %c32768_i64 : i64
            %8 = arith.remsi %7, %c2_i64 : i64
            %9 = arith.divsi %6, %c65536_i64 : i64
            %10 = arith.addi %8, %9 : i64
            %11 = arith.trunci %10 : i64 to i32
            %12 = arith.addi %arg8, %11 : i32
            affine.yield %12 : i32
          }
          affine.yield %1 : i32
        }
        affine.store %0, %arg2[%arg4 + %arg3 * 12] : memref<144xi32, 1>
      }
    }
    return
  }
```

The lowering command I am using is:
```
$mlir-opt --lower-affine <input>-o <output> 

$mlir-opt --scf-for-to-while <input>-o <output> 

$circt-opt --lower-scf-to-calyx --canonicalize --verify-diagnostics <input> -o <output> 

$circt-translate -export-calyx --verify-diagnostics <input> -o $2/$3_calyx_naive.mlir

$calyx <input> --log info -p validate -p discover-external -p validate -x discover-external:default=DEPTH -x discover-external:strip-suffix=_ -o <output> 

$calyx -l $Calyx_PATH -b verilog --synthesis --disable-verify --nested <input> -o <output> 

$calyx -l $Calyx_PATH -b xilinx --synthesis --disable-verify --nested <input> -o <output> 

$calyx -l $Calyx_PATH -b xilinx-xml --synthesis --disable-verify --nested <input>-o <output> 
```

rachitnigam, 2023-10-25:
Thanks for opening this! As I mentioned, the right solution is implementing https://github.com/llvm/circt/issues/4831 but in case this becomes a blocker, let me know @zzy666666zzy and I can try to fix it with even more hacks