calyxir/calyx #1767: Overly aggressive compaction

bcarlet, 2023-11-09:
Compiling the following component through the `pre-opt` stage results in a `static<1> par` block that runs both `invoke`s in parallel.

```
import "primitives/core.futil";

component example() -> (out: 8) {
  cells {
    r0 = std_reg(8);
    r1 = std_reg(8);
    add = std_add(8);
  }
  wires {
    add.left = r0.out;
    add.right = 8'd1;
    out = r1.out;
  }
  control {
    seq {
      invoke r0(in = 8'd1)();
      invoke r1(in = add.out)();
    }
  }
}
```

Assuming this code is legal, the expected behavior is for the `invoke`s to run sequentially due to the data dependency through continuous assignments. Disabling `schedule-compaction` resolves the issue.

Incidentally, given the following component and the example above, the `invoke ex()()` is compiled into a `static<2> group` despite `example` being compiled into a `static<1> component`, so `main` ends up being `static<3>`.

```
component main() -> () {
  cells {
    @external mem = std_mem_d1(8, 1, 1);
    ex = example();
  }
  wires {}
  control {
    seq {
      invoke ex()();
      invoke mem(addr0 = 1'd0, write_data = ex.out)();
    }
  }
}
```

rachitnigam, 2023-11-09:
Continuous assignments strike again! CC @calebmkim @paili0628

calebmkim, 2023-11-09:
Yeah, a simple solution would be something like this: 
- If a group reads from a cell that is written to in a continuous assignment, then its automatically depends on all of the groups before it in the seq. 
- Otherwise we can compute dependencies normally. 

Does that sound like the right solution?

rachitnigam, 2023-11-09:
Do you think this might need to stretch to entire sub-programs? That is, if there a group that accesses a cell, the entire subprogram (`seq` or `par`) is now "infected" and needs to be sequential w.r.t. to everything else?

rachitnigam, 2023-11-09:
Also, note that there are two bugs here that @bcarlet highlighted and we need to fix both.

paili0628, 2023-11-09:
> Do you think this might need to stretch to entire sub-programs? That is, if there a group that accesses a cell, the entire subprogram (`seq` or `par`) is now "infected" and needs to be sequential w.r.t. to everything else?

I think this is exactly the case, essentially we are unable to infer dependence relationships if a group reads from a cell written in continuous assigns.

rachitnigam, 2023-11-09:
Right, but that doesn't mean that we can't compact the program before it. We just need to ensure that this group (that uses a cell from continuous assignments) is sequential w.r.t. to all its dependencies.

paili0628, 2023-11-09:
The harder thing to detect is if two groups in a seq "write" to the same cell inside a continuous assignment, since we allow multiple assignments to write to the same cell if the guards are mutually exclusive. Although this is also quite doable when you think about it.

calebmkim, 2023-11-09:
> Do you think this might need to stretch to entire sub-programs? That is, if there a group that accesses a cell, the entire subprogram (seq or par) is now "infected" and needs to be sequential w.r.t. to everything else?

@paili0628 correct me if I'm wrong, but I think we already do this. In other words: 
```
static seq {
  A; 
  static par {...}
  B; 
} 
```
- Any `static seqs` within `static par` will be compacted first (i.e., we compact bottom up). 
- Then, when analyzing the parent `static seq` , the `static par` is treated as a big blob, in which we record all of the reads and writes within it. 

So I really think it's just a matter of detecting when `A`, `B`, or anything inside `static par` reads and writes from continuous assignments. If `A` reads or writes from a cell through a continuous assignment, we just count it as a read/write of that cell. 
 I think the solution may be augmenting `ReadWriteSet` to include a more complex analysis that takes continuous assignments into account.

rachitnigam, 2023-12-12:
@calebmkim @paili0628 was this issue fixed during the deadline push?

calebmkim, 2023-12-13:
Unfortunately I don't think so. It should be on the to-do list.

rachitnigam, 2023-12-13:
Okay, I think we should fix this sooner rather than later because it might potentially affect paper results

PR #1847 (2024-01-12): Compaction Accounts for Continuous Assignments 
Fix #1767. Adds analysis so that compaction accounts for continuous assignments.

The analysis here is conservative, since it essentially treats all the continuous assignment as one big "block". In other words, any time a group affects the continuous assignments, then it will depend on any previous groups that affect on continuous assignments too. For example say we have
```
seq {
  A; // Affects continuous assignments
  B; 
  C; // Affects continuous assignments
  D; 
  E; // Affects continuous assignments
} 
``` 
Then we detect an `A->C->E` dep [... cut]
closed 2024-01-12 by calebmkim