calyxir/calyx #1839: Test failing: insufficient-params

ethanuppal, 2024-01-08:
## installation
I have a M2 MacBook Air.
```sh
git clone https://github.com/cucapra/calyx.git
cargo build
brew install jq
cargo install runt --version 0.4.0 # the latest is 0.4.1 which doesn't work
```
## testing
```sh
$ runt -i core
✗ [core] errors:tests/errors/insufficient-params.futil
 240 passing / 1 failing / 0 missing / 0 skipped / 0 remaining
 ```
## version
```sh
$ ./target/debug/calyx --version
Calyx compiler version 0.6.1
Library location: /Users/ethan/.calyx
```

rachitnigam, 2024-01-08:
Can you run `runt -i core -d` so that it tells you why the test output didn't match?

ethanuppal, 2024-01-10:
Oddly, it appears to have gotten worse.
```
$ runt -i core -d > log.txt
```
[log.txt](https://github.com/calyxir/calyx/files/13895309/log.txt)

sampsyo, 2024-01-11:
I recognize this problem. :zany:

```diff
- extern "<ROOT>/calyx/tests/passes/canonical/dummy.sv" {
+ extern "/Users/ethan/Documents/GitHub/calyx_git_clone_original_source/tests/passes/canonical/dummy.sv" {
```

Here's the problem: those emitted `extern` statements in compiled Calyx code contain absolute paths, which of course differ on everybody's machine. We have a hack in our test harness to work around this by replacing the machine-specific part of the path with the magic string `<ROOT>`:
https://github.com/calyxir/calyx/blob/94dd5b031b2c00d7731518cfd1f380db4fbd3382/runt.toml#L12

However, this hack depends on your Calyx checkout's directory name being `calyx`, verbatim. So if you were to rename `calyx_git_clone_original_source` to `calyx`, this would work.

I have never had any brilliant ideas for how to remove this hack (e.g.., to practically have the compiler output relative paths), but ideas are always welcome…

sampsyo, 2024-01-11:
Separately, what you're seeing for the `insufficient-params.futil` test is a Rust panic:

```
thread 'main' panicked at 'Failed to add primitive.: Malformed Structure: Invali ...
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
fatal runtime error: failed to initiate panic, error 5
```

As the error message suggests, it would be great to try running this command manually with backtraces enabled to see what's going wrong!

ethanuppal, 2024-01-12:
> I have never had any brilliant ideas for how to remove this hack (e.g.., to practically have the compiler output relative paths), but ideas are always welcome…

Likely I don't understand, but if you're trying to deal with relative paths when you don't want to install in some centralized `/usr/lib`-esque location, can't we do what C/C++ tooling does and have some environment variable akin to `CPATH` or flag akin to `-I`?

rachitnigam, 2024-01-16:
You can do `runt -n tests/errors/insufficient-params.futil` to get the exact command that the test runs. Then you can try running the command and playing around with it to see what's wrong.

This bug is probably not reproducible for us because the test passes on the CI. My best guess is that there is something different about Rust versions.

sgpthomas, 2024-01-16:
I ran into something similar while trying to get a new install up and running. You're right about the rust version, looks like panic messages are not stable across rust versions. I'm on `1.75.0`. I guess a short-term solution would be to pin a rust version with a [`rust-toolchain.toml`](https://rust-lang.github.io/rustup/overrides.html#the-toolchain-file) file. A maybe more robust solution is to not panic in the builder so that we can control the stability of the error messages.

```
✗ [core] errors:tests/errors/insufficient-params.futil
         ~
    2   2│  101
    3   3│  ---STDERR---
    4    │- thread 'main' panicked at 'Failed to add primitive.: Malformed Structure: Invalid parameter binding for primitive `std_fp_div_pipe`. Requires 3 parameters but provided with 1.', calyx-ir/src/builder.rs:224:14
        4│+ thread 'main' panicked at calyx-ir/src/builder.rs:224:14:
        5│+ Failed to add primitive.: Malformed Structure: Invalid parameter binding for primitive `std_fp_div_pipe`. Requires 3 parameters but provided with 1.
    5   6│  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
    6   7│  
         ~
```

rachitnigam, 2024-01-17:
Yeah, this has been a consistent problem unfortunately. Not sure what the best solution is here because parameter resolution happens in the AST -> IR conversion step which doesn't bubble up errors. It might be possible to define a version of the method that returns `Result`.

sampsyo, 2024-01-22:
Yes, indeed! It seems like we can either make that change as @rachitnigam outlines or add a separate AST-level check for these problems (so the subsequent conversion is guaranteed to succeed).

PR #1876 (2024-01-28): Dont panic if insufficient parameters provided
Fixes #1839.
closed 2024-01-28 by rachitnigam