calyxir/calyx #1852: Compaction Messes Up Static Promotion

calebmkim, 2024-01-17:
(This was originally mentioned by @bcarlet in #1767). 

Consider the following example component, where `r0` and `r1` are registers: 
```
component example() -> (out: 8) {
  ... 
  control {
    seq {
      invoke r0(in = 8'd1)();
      invoke r1(in = 8'd5)();
    }
  }
}
```

Compaction changes the control into: 
```
static<1> par {static invoke r0; static invoke r1}
```
thereby making example's signature
```
static<1> component example() -> (out: 8)
```

However, suppose we have 
```
component main() -> () {
  cells {
    ex = example();
  }
  ...
  control {
      invoke ex()();
  }
}
```

Since `static-promotion` runs _before_ `schedule-compation`, we will convert this to: 
```
component main() -> () {
  cells {
    ex = example();
  }
  ...
  control {
      static<2> invoke ex()(); // this is correct, but after compaction runs, it should be static<1>. 
  }
}
```
This will be incorrect after compaction runs. 

### Solution
I think the solution is to merge the`static-promotion` and `schedule-compaction` into a single pass. This (hopefully) won't be too hard, since a lot of `compaction` is done in external analyses. I want to wait until after we've deprecated `@static` (#1429) before we do this though, because I want to make sure everyone is on the same page about what the `@latency` attribute is, what `static invoke` actually means, etc. I have a decently clear idea in my mind, but we can discuss this in the next meeting.

rachitnigam, 2024-01-17:
Sorry, why do we need to merge passes? Can't compaction go back and update the invokes? It will require recomputing the latencies of components but that doesn't seem too bad (i.e. that logic already exists in static-promotion; we might need to refactor it into an analysis or something)

calebmkim, 2024-01-17:
We definitely don't have to merge the passes; your solution works too. There were a couple reasons why I think it might be nice (although definitely not necessary) to merge the passes. 

1. We can make decisions on whether to promote based on the compacted control schedule, rather than the un-compacted schedule. For example, if we have `if port { seq1} else {seq2} ` suppose `seq1` originally takes 5 cycles and `seq2` originally takes 10 cycles. We might not want to promote it, since the `static if` would conservatively take 10 cycles. But then, we find out we can compact `seq2` to take 5 cycles but we cannot compact `seq1`. Then, we might want to promote it, based on this new information about `seq2`. 
2. When `compaction` goes back and updates the invokes, it will also have to update any groups that use the component as well. Then, it will have to propogate the those changes "upwards" in the "control tree" (e.g., we will have to retime a `seq` whose child invokes the component). It seems like we would basically be re-doing `static-promotion` again.
3. (Not all that important) It was a bit weird that we are allowed to reschedule a `static seq`.  It would be nice not to have that imo. 

Either way though, I think we may have a better intuition about which solution is correct after I try to implement it.

rachitnigam, 2024-01-19:
Oh interesting! This almost makes me think that compaction should be operating on `@promote` marked control programs instead of `static` programs. The point about `if` branches is particularly salient because it means that our threshold for decisions might change as we compact schedules, which in turn might enable use to compact even more!

I think the heuristics would become more complex because we'd be reason about chains of transformations likes: "if I promote this child, compact this branch, it will enable this other child to be promoted under the current thresholds". We should be careful when structuring this pass and its code.

A meta-point is that this, once again, the phase ordering problem at play. `egg` style approaches would make something like this trivial because we could saturate the program with all possible compaction and promotion rewrites and pick the best one.

Anyways, I'm more open to merging the two passes if it enables this kind of optimization

sampsyo, 2024-01-22:
> This almost makes me think that compaction should be operating on `@promote` marked control programs instead of `static` programs.

Yes! This is, at least, the "morally" correct thing to do. That is, if you believe that `static` programs are supposed to come with timing guarantees, they should prevent any kind of reordering of execution (which compaction does). So yes, I very much endorse the idea of making compaction work on annotated (as opposed to qualified) code!

But of course, merging the two passes obviates the issue as well, because then the combined pass isn't "disobeying" any actual IR commandments.

Super interesting issue about phase ordering, in general… it does seem like it argues for either (a) stronger heuristics, or (b) some kind of weirdo equality saturation approach.

calebmkim, 2024-02-01:
Closed by https://github.com/calyxir/calyx/pull/1884

closed 2024-02-01 by calebmkim