calyxir/calyx #1865: AXI generator clean up

nathanielnrn, 2024-01-23:
A few small issues that were brought up that should be addressed. I find it helpful to keep these all in one place.
1. As mentioned [here](https://github.com/calyxir/calyx/pull/1854#discussion_r1461248966), in our channels we create `base_addr` and `curr_addr` registers. The naming is a bit vague and `base_addr` should be changed to something that reflects that this goes to our byte-addressed AXI addresses, while `curr_addr` should be changed to reflect that it indexes into our internal calyx memories based on the memories indexing.
2. Assert in `build` that our `yxi` is well formed. i.e, [powers of 2](https://github.com/calyxir/calyx/pull/1856#discussion_r1461253809), width is byte aligned, etc.
3. Reduce code duplication by refactoring where appropriate.
4. Talked about this synchronously with @sampsyo and have had a few more thoughts since then:
Reevaluate `xVALID_was_high` signals based on our in sync discussion. In general the point of this is to denote we are not in the first "raising of `xVALID`" cycle and that a handshake has occurred. We need this because we want to ensure that in a single servicing of a transfer `xVALID` is only asserted once. Because we have no guarantees on when the servicing group will terminate, we look to `xVALID_was_high` to ensure xVALID is only high once. When we discussed this in person we assumed this would be in the first cycle, but I think this should be generalized to denote when a `handshake_occurred`. More concretely if we are in the group that services a transfer I believe we want the following (being a bit sloppy with notation):
    ```
    //we are not currently in a handshake and haven't been high before
    xVALID.in = !(xVALID & xREADY) & !(handshake_occurred) ? 1'b1;
    //negate the above. Key point is that `handshake_occurred` stays high.
    xVALID.in = (xVALID & xREADY) | handshake_occurred ? 1'b0;
    xVALID..write_en = 1'b1;
    
    handshake_occurred.in = (xVALID & xREADY) ? 1'b1;
   // only write once on first handshake
    handshake_occurred.write_en = (xVALID & xREADY) & !(handshake_occurred) ? 1'b1;
    ``` 
    
   I spent some time trying to think about if this could/should be reduced anywhere, but I think this is what is necessary to get our implementation to work.

sampsyo, 2024-01-26:
This sounds perfect. That new definition for `handshake_occurred` crystalizes what's going on here: we want to keep track of the moment when the handshake happened, by that natural "valid && ready" definition, and use that to suppress future handshakes.

nathanielnrn, 2024-02-16:
Closing this for now, leaving the refactoring item open as something that can be done with a later issue.

closed 2024-02-16 by nathanielnrn