calyxir/calyx #1873: Using fud priority causes warning

sgpthomas, 2024-01-26:
I wanted to make verilator the default verilog simulator for fud, so I set the `stages.verilog.priority` key to 1. This works, but causes a warning saying that the key is unknown. I don't think that this should generate a warning. 

The warning happens because the Verilog stage doesn't define it as a known key in the `known_opts` function. It seems silly to have to add the priority key for every stage, so maybe there's a way we can always have this be a `known` opt for all stages.

Of course this is not a big deal at all. And maybe `fud2` will just supercede `fud` and this is not worth fixing.

rachitnigam, 2024-01-27:
Argh that's odd! The logic for checking is here and we can change it to not check the `priority` key: https://github.com/calyxir/calyx/blob/main/fud/fud/stages/__init__.py#L218

However, it's kind of weird that we use the stage options magic to encode priority stuff anyways. `fud2` should probably make a better trade-off here (CC @sampsyo).

sampsyo, 2024-01-27:
Thanks for the ping! I haven't implemented anything like priority in fud2 yet. Seems worth doing. Currently, the first thing that the BFS algorithm finds "wins" unless you use `--through`! Food for thought… I also don't know what the right UI would be for this yet.