calyxir/calyx #1878: fud2 Tracker

sampsyo, 2024-01-28:
Here is a tracking issue to catalog all the stuff to do to fud2, after the initial import in #1877. As I turn these bullet points into issues, I'll edit this issue to link to them.

Near-term infrastructure completion:

* [ ] Tests! Maybe unit tests, or maybe snapshot tests of the emitted Ninja stuff. (Actual functionality tests of real builds will happen elsewhere, i.e., when our ecosystem-wide tests eventually use fud2 as a fud alternative.)
* [ ] Document all the currently supported ops.
* [ ] Document the design (and the scope/goals of both fake and fud2), and how to add functionality to fud2.
* [ ] Add a CLI option to list all the states & ops.
* [ ] Better error messages w/r/t configuration problems.

Missing functionality from OG fud (an incomplete list):

* [ ] XRT: Make trace mode (VCD emission) work.
* [ ] XRT: Support real-FPGA execution mode.
* [ ] Systolic array generator.
* [ ] Vivado HLS.
* [ ] Vivado report parsing?

Big features to explore:

* [ ] Extensibility. Perhaps with a scripting language like Starlark, as outlined [elsewhere](https://github.com/orgs/calyxir/discussions/1837#discussioncomment-8024380).
* [ ] Ops with multiple inputs. The point is that the `--set sim.data=foo.json` route, where the data input is a "second-class citizen" w/r/t the code simulation, is a hack. We should instead treat `foo.json` as a proper input, just like the Verilog program, that goes through its own op-based transformation, discovered with a BFS traversal of the op graph. This would let you, for example, do `fud2 foo.json -o foo.hex` or whatever to run the data conversion alone, and it would allow other formats other than the "blessed" JSON format to work transparently.
* [ ] Ops with multiple outputs. This is relevant mostly to the simulation stages… currently, we have separate `icarus` and `icarus-trace` ops because one produces a JSON data file and the other produces a VCD file. Maybe we can make this one op? Then you wouldn't have to do `--through icarus-trace --to vcd`, which seems redundant; it would suffice to just do `--through icarus --to vcd` to specify which output you want.
* [ ] Remote execution. We had this in OG fud for the Xilinx ops, but I think this should be a generic thing that any op can use.
* [ ] Priority. This is an OG fud feature we don't have. See #1873.
* [ ] "Check mode." OG fud has `fud check` that can automatically diagnose lots of installation and versioning problems.
* [ ] "Install mode." Like check mode but even more helpful. See https://github.com/sampsyo/fake/issues/2.
* [ ] Rethink how "support resource" files work. Lots of ops need auxiliary files, and we have to deliver those somehow. Currently, those go in a directory alongside the actual code for fud2. This is inelegant because it means fud2 is not a self-contained executable, and fud2 isn't usable until you set a config option to point to that directory. Maybe we should [include the files in the binary itself](https://github.com/magiclen/lazy-static-include).
* [ ] (Optionally) embed [Turtle](https://github.com/raviqqe/turtle-build) instead of relying on Ninja?

rachitnigam, 2024-01-29:
Thanks for getting started on this @sampsyo! What do you think about the following milestone: getting all of the tests in the monorepo working with `fud2`? We can maintain a separate `runt.toml` file and slowly etch away at all the failures/missing features. Once we have that done, we can consider recommending `fud2` as the default build tool and see what other things keel over?

rachitnigam, 2024-02-06:
> Ops with multiple inputs

I came across when playing with `fud2` as well; there is some feeling about treating inputs more declaratively in the same way we think about files. The idea that some secondary inputs need to be transformed themselves is already true for things like the symbolic evaluator where the input spec might need to compiled through its own `fud` flow.

rachitnigam, 2024-02-06:
Can we add https://github.com/calyxir/calyx/issues/1899 to the tracker @sampsyo?

sampsyo, 2024-02-10:
Yes, having "secondary inputs" be treated as first-class citizens would be really really satisfying…

rachitnigam, 2024-02-16:
Proposal: `fud2 package` to create zip/tarballs containing all of the source files needed to execute a particular flow. This could be useful for remote workflows and particular powerful for benchmarking infrastructures that want to generate one ninja file to generate results for a bunch of different input programs.

rachitnigam, 2024-02-16:
> Several ops currently depend on chunks of Python functionality from OG fud. We should figure out what to do about this in general… maybe just document it? Or create a separate Python package you have to install alongside our Rust business? Or just replace those bits of functionality altogether, which sidesteps the headache.

I think this should be promoted into a general discussion about a new `tools/` folder and we should formalize the process of building new tools for Calyx (languages supported, build flows, especially for python).

rachitnigam, 2024-02-19:
One problem I've been running into when using `fud2` is if the Calyx compiler itself change, the `.fud2` is not invalidated and `fud2` will skip the compilation step. We should add a dependency on the various tools being run to make sure that when the tools themselves change, the build is re-executed from the right step!

Woo! New set of challenges from incremental execution but it'll be worth it once we have it figured out.

rachitnigam, 2024-05-09:
This is a cool idea!! A potential challenge is that this make `fud2` a full blown build tool (like `fud`). For example, the approach of "fusing" build steps implies that ninja does not get to see (and cache) the results of build artifacts. This also complicates the debugging story; there are now two possible flows for every build: one through `fud2` and one through `ninja`.

An alternative approach to the serialization problem is coming up with binary formats that can be `mmap`d directly by the various tools. For example, we don't really have a binary format for Calyx IR which causes expensive serialization and deserialization.

sampsyo, 2024-05-13:
Ah, yes, thank you for articulating the downside. It would cost a lot of complexity! And, as you say, create two implementations to test for every possible plan. I guess one thing it wouldn't incur, however, is process management: everything would be in-process (that would be the point) so no `std::process::Command` necessary.

And yeah, you're right that there might be better alternatives to explore if serialization becomes a bottleneck… I guess the real lesson here is that we should wait for an actual bottleneck to appear before picking the solution. 🤪