calyxir/calyx #1889: Allow for addition to `par` control blocks in builder

nathanielnrn, 2024-02-02:
Following from some discussion in the [slack](https://cucapra.slack.com/archives/CLLM75W4R/p1706883742201499) it seems like it would be useful to allow for following construct in the builder:

```
ctrl.par(A)
```
will output
```
par {
  ctrl;
  A;
}
```

To that end it might make sense to make par similar to the current [`__add__`](https://github.com/calyxir/calyx/blob/cc6783e084bc20fb425bbfd0dcb6d48e72db9a76/calyx-py/calyx/builder.py#L897-L915) function, copied below for reference.

```python
    def __add__(self, other):
        """Build sequential composition."""
        other_stmt = as_control(other)

        if isinstance(self.stmt, ast.Empty):
            # Special cases for when one side is empty.
            return ControlBuilder(other_stmt)
        elif isinstance(other_stmt, ast.Empty):
            return self
        elif isinstance(self.stmt, ast.SeqComp) and isinstance(other_stmt, ast.SeqComp):
            # Special cases for when we already have at least one seq.
            return ControlBuilder(ast.SeqComp(self.stmt.stmts + other_stmt.stmts))
        elif isinstance(self.stmt, ast.SeqComp):
            return ControlBuilder(ast.SeqComp(self.stmt.stmts + [other_stmt]))
        elif isinstance(other_stmt, ast.SeqComp):
            return ControlBuilder(ast.SeqComp([self.stmt] + other_stmt.stmts))
        else:
            # General case.
            return ControlBuilder(ast.SeqComp([self.stmt, as_control(other)]))
```

Perhaps combining this new `par()` function with some refactoring that would allow us to abstract the idea of "unwrapping if both arguments are the same/first arg is something/second arg is something."

sampsyo, 2024-02-04:
Sounds exactly right to me!!

nathanielnrn, 2024-03-03:
As discussed in the (not merged) [PR](https://github.com/calyxir/calyx/pull/1894#issuecomment-1953092782), this is not currently planned as there are ways around this. Namely, constructing lists and and unpacking them accomplishes the same thing this was aiming to do, without adding all sorts of assumptions we expect clients of the builder to know about.

closed 2024-03-03 by nathanielnrn