calyxir/calyx #1913: Cider 2.0 Tracker Issue

EclecticGriffin, 2024-02-12:
We're finally at point where we can start running programs through Cider 2 and thus it now makes sense to have a tracker for the ongoing efforts to reach parity with Cider 1.

Known issues:
- [ ] Does not generate an output JSON compatible with our testing infrastructure
- [ ] Missing a bunch of primitives
  - [ ] Comb primitives are partially done
  - [ ] Seq Mems
  - [ ] Mult/Divs
  - [ ] Fxp 
- [ ] Group termination takes an extra cycle due to only running convergence once. Issue: TK
- [ ] `par` not supported
  - [ ] cross branch conflicts warnings missing
- [ ] `with` not supported
- [ ] `invoke` not supported
- [ ] multi-component programs not supported
- [ ] all the helpful info logging & error messages

rachitnigam, 2024-02-13:
For some of these issues, like `with` not being supported, we should resolve the long-running `with` discussion and decide if we want to get rid of it completely or not.

EclecticGriffin, 2024-02-15:
@rachitnigam Agreed, this is mostly just me getting down the current state of things. Also `with` is annoying to implement so I've thus far avoided it oops

sampsyo, 2024-02-27:
Wahoo; thank you for summarizing! Just capturing the synchronous discussion a little bit, on this point:

> Does not generate an output JSON compatible with our testing infrastructure

I had suggested, in the spirit of trying to do the simplest possible thing here, that it might not be too hard to load/dump "raw bits," i.e., the exact bit-level contents of the memories. This is more or less what the RTL simulators already do (they are using hex-encoded text, but same difference). We could then pre-/post-process these files into our JSON format externally, alleviating the need for any `serde` hacking on Cider's side.

This of course omits all the non-memory results that Cider 1.0 can already produce. But I believe this is fine: all we really need to check correctness is those memory dumps.

EclecticGriffin, 2024-08-06:
Officially closing this tracker since at this point Cider 2 has replaced Cider 1 and has passed parity in terms of programs

closed 2024-08-06 by EclecticGriffin