calyxir/calyx #1916: fud dependency on calyx-py

eys29, 2024-02-14:
I think this recent [push adding an import from calyx](https://github.com/calyxir/calyx/blob/a2f2f3cb51c8e65209df6b85c0a02eae98f8afed/fud/fud/stages/verilator/json_to_dat.py#L3C1-L4C1) reintroduces the issue [here](https://github.com/calyxir/calyx/pull/597/commits/f26e35343f38ee8b2572e9fbedb1d2ad4b6a0915#diff-284a7711258a58ecff116cb6c13d95fa59f3b3b78670d12d46afdf23b3bb3dc5). 
I was having trouble reinstalling fud and was following this [thread] (https://cucapra.slack.com/archives/CLLM75W4R/p1631711896069400). 

The command `fud config --create global.root <full path to Calyx repository>` would complain about `import calyx` in  [json_to_dat.py](https://github.com/calyxir/calyx/blob/a2f2f3cb51c8e65209df6b85c0a02eae98f8afed/fud/fud/stages/verilator/json_to_dat.py#L3C1-L4C1) and [interpreter.py](https://github.com/calyxir/calyx/blob/a2f2f3cb51c8e65209df6b85c0a02eae98f8afed/fud/fud/stages/interpreter.py). 

I tried removing the `import calyx` lines in these two files on this [branch](https://github.com/calyxir/calyx/tree/fud-import-calyx) which allows the `fud config` command to run but then `fud -e` obviously can't import the numeric types.

eys29, 2024-02-15:
cc @calebmkim @rachitnigam

rachitnigam, 2024-02-16:
@eys29 CAPRA slack links are not accessible from this page. Can you update the issue to be self-contained? 

Also, this happened because of a recent merge (#1715). You need to install `calyx-py` to make `fud` work now (which is kind of annoying I guess): https://docs.calyxir.org/builder/calyx-py.html#installation

eys29, 2024-02-16:
Thanks @rachitnigam for pointing that out! Installing calyx-py fixed that.

closed 2024-02-16 by eys29