calyxir/calyx #1918: Create an `external-to-ref` pass

nathanielnrn, 2024-02-16:
To ease the integration of Calyx programs into our AXI-wrapper it would be helpful to have an `external-to-ref` pass that changes toplevel memories marked `@external` to `ref` memories. I believe this would require changing part of the wellformedness check, which doesn't currently allow for `ref` in the toplevel? Unless we want to limit this pass to cases where we're using `--synthesis` flags.

Hopefully this would also help with any future work to deprecate `@external` in favor of `ref`, as discussed [here](https://github.com/calyxir/calyx/issues/1603#issuecomment-1937020425).

nathanielnrn, 2024-02-16:
I will also say that this would be my first pass! So any pointing-in-the-right-direction would provide a big head start on this.

sampsyo, 2024-02-27:
Yes, let's change that well-formedness check! IMO we should just disable it, as part of the `@external` deprecation effort (cf https://github.com/calyxir/calyx/issues/1603#issuecomment-1937020425). `--synthesis` is also headed for the grave.

nathanielnrn, 2024-03-03:
Closed by #1952

closed 2024-03-03 by nathanielnrn