calyxir/calyx #1961: `clk` signal not respecting cycle length

paili0628, 2024-03-06:
On the current `improve-comb-prop` branch, when I generated the vcd waveform for the failing test case `tests/correctness/invoke-memory.futil`, it showed the waveform below: 
[image]
where the clk signal continuously oscillated up and down instead of respecting the cycle length.

The command I ran is `fud exec tests/correctness/invoke-memory.futil -o out1.vcd -s verilog.data tests/correctness/invoke-memory.futil.data --from calyx -s verilog.cycle_limit 500`.

rachitnigam, 2024-03-06:
When I zoom in, I get the expected waveform. Can you explain what you mean by "signal continuously oscillated up and down instead of respecting the cycle length"?

paili0628, 2024-03-06:
This is what I get when running the same command on the main branch:
[image]
The cycle length is much longer with a correctly running instance of the same calyx program `tests/correctness/invoke-memory.futil`. 
I guess my question is might there be a possible reason that is causing the abnormal cycle length.

rachitnigam, 2024-03-06:
Okay, so the design is just running for longer which means that the `done` condition is optimized away in a way that makes it wrong. This doesn't have anything to do with the `clk` signal which is pulsing as it is meant to.

rachitnigam, 2024-03-14:
@paili0628 I'm going to close this. Let us know if you're running into similar problems.

closed 2024-03-14 by rachitnigam