calyxir/calyx #1966: Default nettype of inline verilog primitives is unset

matth2k, 2024-03-10:
Can we explicitly set the `default_nettype` directive for inlined Verilog primitives? Most cases we probably want to set it to `wire`. It is causing issues when trying to synthesis the output of `calyx -b verilog --synthesis` with Vivado. 

To elaborate, the `core.futil` primitives are the last thing to update the default nettype:
https://github.com/calyxir/calyx/blob/bdbddcb3edf09d76b6d3a121ea915832d0774f2b/primitives/core.sv#L9-L12
These are included before `compile.futil` which are printed to verilog without an explicit type:
https://github.com/calyxir/calyx/blob/bdbddcb3edf09d76b6d3a121ea915832d0774f2b/calyx-backend/src/verilog.rs#L178-L190

As a result, I get errors on things like `std_add` and my own external components with Vivado.

```
ERROR: [Synth 8-6735] net type must be explicitly specified for 'in' when default_nettype is none [/work/shared/users/phd/mrh259/calyx/vivado/benchmark.srcs/sources_1/imports/mrh259/calyx/benchmark.sv:602]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'left' when default_nettype is none [/work/shared/users/phd/mrh259/calyx/vivado/benchmark.srcs/sources_1/imports/mrh259/calyx/benchmark.sv:611]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'right' when default_nettype is none [/work/shared/users/phd/mrh259/calyx/vivado/benchmark.srcs/sources_1/imports/mrh259/calyx/benchmark.sv:612]
```

calebmkim, 2024-03-14:
Thanks for looping me in to this issue. 

What would the solution to this be? Would it basically be: allowing inlined primitives to parse `default_nettype <any_string>` and then print `default_nettype <any_string>` (i.e., the same string) right before we print the inlined primitives in the Verilog backend?

matth2k, 2024-03-14:
My personal pick on what to do would be to just not to use the `default_nettype` directive. Then, try not to use any implicit declarations of nets. So remove `default_nettype none` from core.sv. Then I would adjust the Verilog emitter to write `output wire` for each output of the primitive.

This way, we try not to have any implicit types. But we don't crash if we do. I could draft up a PR later if we think this the right idea.

rachitnigam, 2024-03-14:
Sounds like the right idea to me!