calyxir/calyx #1968: Does not generate an output JSON compatible with our testing infrastructure

EclecticGriffin, 2024-03-11:


EclecticGriffin, 2024-03-12:
I am currently leaning toward having a simple preamble with this info since I think it will make things simpler in the long run, but I am open to other ideas. CC: @sampsyo

sampsyo, 2024-03-12:
All sounds awesome. I agree with the decisions you summarized briefly:

* little endian
* pad to bytes
* row-major for multi-dimensional memories

And very good reasoning about the metadata that describes what's in the file(s), which of course is necessary for producing any other format from such a binary dump. I actually think this dovetails nicely with two other things going on in the ecosystem.

First:

> it may be worth having a preamble which contains the names of the memories, their dimensions, and the definition order

This is pretty much the goal of the "YXI" interface definition format created by @nathanielnrn for AXI purposes! Check it out:

```
$ calyx -b yxi examples/futil/dot-product.futil
{
  "toplevel": "main",
  "memories": [
    {
      "name": "A0",
      "width": 32,
      "size": 8
    },
    {
      "name": "B0",
      "width": 32,
      "size": 8
    },
    {
      "name": "v0",
      "width": 32,
      "size": 1
    }
  ]
}
```

That is, YXI is just a JSON document that includes the names and dimensions of the exposed top-level memories (using `@external` or `ref`). It's meant to be a comprehensive description of the "external interface" to a Calyx program. So maybe it is exactly what we want here?

So, Cider 2.0 could dump "just the bytes" and rely on a separate YXI file to interpret it. Or, it could produce a single file that consists of the YXI data (presumably serialized to some other format) followed by all the bytes. Either way, maybe it would be cool to standardize on YXI being the way to describe this stuff, extending it if necessary to address this use case (as opposed to inventing a new/different format with similar-but-not-quite-identical contents)?

> and have a tool binary, in the fud2 philosophy, which can convert json to memory dumps and vice-versa

I don't think I've broadcasted this too broadly, but @bcarlet and I have recently started working with @Angelica-Schell to do something like this!! That is, we are starting small, but we are hoping to build a standalone data converter tool (as you say, taking the fud2 approach) that can convert between many different data formats. Including Verilog-simulator-friendly binary files, OG dud-style JSON, anything else we can think of. And Cider's preferred format could be wrapped up into that!

Anyway, this is just to say that we should totally build such a thing, and it should probably be a command-line flag tacked onto what @Angelica-Schell is beginning to construct now.

EclecticGriffin, 2024-03-12:
> don't think I've broadcasted this too broadly, but @bcarlet and I have recently started working with @Angelica-Schell to do something like this!! That is, we are starting small, but we are hoping to [...]

Ah brilliant, that's exactly what I was thinking about. Happy to help out if needed.

> This is pretty much the goal of the "YXI" interface definition format created by @nathanielnrn for AXI purposes!

Amazing! How does this look for multidimensional memories with the size definition?

Essentially what I was imagining was basically the yxi interface info (serialized into a binary format) followed by the raw binary data for all the memories one after another. I think that is better than the version where we have the header for a single memory followed by the data for that memory and so on, since we can easily extract the header info from that without needing to parse the entire file.

sampsyo, 2024-03-12:
I don't remember of the top of my head where we landed on multi-dimensional memories, but IIRC we either (a) don't handle them at all, or (b) just report the total size (i.e., the product of the dimensions). And glancing quickly at the code, I think it's (b), as in, it uses `get_mem_info`:
https://github.com/calyxir/calyx/blob/23cd0a1be9724663023844756493e3a5044b53c2/calyx-ir/src/utils.rs#L39

> I think that is better than the version where we have the header for a single memory followed by the data for that memory and so on, since we can easily extract the header info from that without needing to parse the entire file.

Yeah, totally makes sense to me.

PR #2041 (2024-05-14): [Cider 2] Memory loading & dumps
closes #1968

Adds `cider-data-converter` tool which transforms the json data into the new cider binary data dump format and back again. This also adds the ability to load/dump memories from/into this format in the flat interpreter. For now the data-converter only works for unsigned integer values of at most 64 bits. In the future, I would like to replace this with some bigint stuff and support fixed-point again but I'm not considering either high priority right now.

The data conversions all have property tests implemented as well!

Also did some rearranging o [... cut]
closed 2024-05-14 by EclecticGriffin