calyxir/calyx #2014: [Profiling] Tracker Issue for Profiling first steps

ayakayorihiro, 2024-04-22:
This issue lists out first steps for profiling! (Mostly so I can organize my TODOs.) Will update as I move along.

### First Pass: Cycle-level performance info at the Calyx level

- [ ] Metadata generation
   - [ ] Obtain JSON from TDCC (add another pass option to print JSON instead of the dump)
- [ ] Loading in the trace
   - [ ] Figure out what tool to use?
- [ ] Make flame graphs
   - [ ] There is probably a library out there for this.

rachitnigam, 2024-04-25:
Thanks for opening this @ayakayorihiro! Could you add the "Tracker" label to this issue?

ayakayorihiro, 2024-04-25:
Thanks @rachitnigam ! Just added the tracker label, will keep in mind for next time :)

ayakayorihiro, 2024-05-13:
Thanks @ekiwi ! I'll take a stab following your work with the `vcdvcd` library :)

ayakayorihiro, 2025-06-23:
Closing this issue since I've been using a different tracking system for my TODOs.

closed 2025-06-23 by ayakayorihiro