calyxir/calyx #2050: CIDR: bug when running `invoke`

anshumanmohan, 2024-05-23:
I am getting 
```
thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', interp/src/primitives/stateful/memories.rs:570:9`
```

When trying to run CIDR on a program featuring an invoke and a memory passed by reference.

I have a minimal example [here](https://github.com/calyxir/calyx/blob/cidr-unwrap-bug/calyx-py/test/correctness/queues/binheap.futil). My exact command was 
```
fud e --to debugger -q binheap.futil -s calyx.flags '-p compile-invoke' -s verilog.data binheap.data
```

Removing the invoke on line 26 solves the issue and starts up CIDR.

EclecticGriffin, 2024-05-23:
Okay so I'll take more of a look tomorrow, but was unable to quickly reproduce it. Can you confirm that your cider binary is up to date?

EclecticGriffin, 2024-05-24:
So I tried again and checked out the branch you put the file in and did a clean build and am unable to reproduce the error. Cider starts in the debugger mode as normal on my machine.

PR #2138 (2024-06-12): Docs: Add fud registration instructions for Interpreter
Closes #2050.

In the [page about the interpreter](https://docs.calyxir.org/running-calyx/interpreter.html#interpreting-via-fud), the docs were not showing people how to register the interpreter as a fud stage. In the [page about the debugger](https://docs.calyxir.org/debug/cider.html#getting-started), the docs were not telling people to build and register the interpreter first. This PR fixes both of these and makes a couple quick clarity changes.

The PR also guides users about the interpreter not supporting the `invoke` command, and exp [... cut]
closed 2024-06-12 by anshumanmohan