calyxir/calyx #2057: Compilation bug: mysterious cross-connection

anshumanmohan, 2024-05-25:
I have found a case where the interpreter gives me the answer I expect, but Verilator and Verilog both give the same incorrect answer. The minimal example is [here](https://github.com/calyxir/calyx/blob/binheap-debug/calyx-py/test/correctness/queues/binheap.futil). I am including the respective outputs, which you must run from `calyx-py/test/correctness/queues`. I've also included the result of running `fud check`.

The offending line is 
https://github.com/calyxir/calyx/blob/f0ececd3d4aeba4cd68a2c5c79f6bc95d0e99ffd/calyx-py/test/correctness/queues/binheap.futil#L36

If you remove it from the control and the group from the wires, all three drivers give the same correct output. I'm pretty confused because `find_parent` is such a non sequitur: the `binheap` component literally just adds the latest item to cell `mem[size++]`. Amid this, `find_parent_idx` just sets `parent` to 0, and does not touch `size`.

### Interpreter (Correct)

```
fud e --to interpreter-out -q binheap.futil -s calyx.flags '-p compile-invoke
' -s verilog.data binheap.data
{
  "main": {
    "mem": [
      9,
      6,
      3
    ]
  }
}
```

### Verilog

```
cat binheap.futil | fud e --from calyx --to jq \
       --through verilog \
       --through dat \
       -s verilog.data binheap.data \
       -s jq.expr ".memories" \
       -q
{
  "mem": [
    3,
    0,
    0
  ]
}
```

### Icarus Verilog

```
cat binheap.futil | fud e --from calyx --to jq \
       --through icarus-verilog \
       --through dat \
       -s verilog.data binheap.data \
       -s jq.expr ".memories" \
       -q
{
  "mem": [
    3,
    0,
    0
  ]
}
```

### `fud check`

```
fud check
global:
 ✔ /scratch/anshuman/calyx exists.

stages.calyx.exec:
 ✔ /scratch/anshuman/calyx/target/debug/calyx installed.

stages.interpreter.exec:
 ✔ /scratch/anshuman/calyx/target/debug/cider installed.

...

stages.verilog.exec:
 ✔ /opt/verilator/bin/verilator installed.
 ✔ Found version 5.006 (>= 5.002).

stages.icarus-verilog.exec:
 ✔ iverilog installed.
 ✔ Found version 11.0 (>= 11.0).
```

anshumanmohan, 2024-05-26:
Ah I forgot to mention another piece of detective work I did. If you set `find_parent_idx` to always return some other hardcoded value, then 6 and 3 are put there instead of 0. The 6 is clobbered by the 3. So if `find_parent_idx` always returned 2, the output from Verilog would be `[9, 0, 3]`.

I tested the clobbering behavior by making the main component invoke the called component only twice, with 9 and 6. When parent-finding is hardcoded to 0, the output is `[6, 0, 0]`. When hardcoded to 2, the output is `[9, 0, 6]`.

rachitnigam, 2024-05-26:
Oh, interesting find! One thing I would try to do is minimizing the example using the [`flag-compare.sh`](https://github.com/calyxir/calyx/blob/main/tools/flag-compare.sh) file. You can set the `fud` commands to be run to perform the output diff you want.

If you're feeling more ambitious, you can try using [creduce](https://github.com/csmith-project/creduce) to automate the whole thing! We've not tried it on Calyx but it might be able to work because we have somewhat C-like syntax.

anshumanmohan, 2024-05-27:
Cool! Also, maybe this was already painfully obvious to you haha, but this is a cell-sharing issue. I just ran the verilog/verilator commands with cell-sharing turned off and got the expected answers. Now I'll try to minimize.

calebmkim, 2024-05-27:
I'll take a look at what's going on with cell sharing

anshumanmohan, 2024-05-27:
I have minimized the example. The top post now discusses and points to the new code.

When trying to minimize, I found some examples of things that make the problem go away:
- Inline the work of `memsto` into `main`, thereby massaging away the invokes.
- Remove the second invoke, and just duplicate `memsto`'s control so that it reads 
```
comp.control += [
        mem_store,
        incr_i,
        zero_out_j,
        mem_store,
        incr_i,
        zero_out_j,
    ]
```

Anyway, because they removed the problem, I have not presented these options in the minimized example. Also, the .py file accurately generates the .futil file.

anshumanmohan, 2024-05-27:
I also checked that the issue wasn't because I was indexing with bitwidth 4, which is perhaps unusual. I now index by 8.

anshumanmohan, 2024-05-27:
Most interesting of all: I decided that if `i` and `j` are just being shared, then why not leave `i` alone and increment `j` instead? So, in pseudocode:

```
reg i;
reg j;
mem[i] = 4;
j++;
```

But this caused the compiler to behave correctly, i.e., it did not share the cells anymore.

It's only when you do what I have in the example:
```
reg i;
reg j;
mem[i] = 4;
i++;
j := 0
```

that the cells get shared.

PR #2071 (2024-05-30): Sharing Bug Fix
Fixes #2057. 

Our analysis assumed that state would not live beyond a single invocation (similar concept to the `infer-share` pass). But as #2057 shows, if a component's state lives beyond a single invocation, then our live range analysis should account for this. 

The solution is to do exactly what we do for `while/repeat` loops.
closed 2024-05-30 by calebmkim