calyxir/calyx #2061: Floating-point constants

rachitnigam, 2024-05-27:
@jiahanxie353 has proposed adding floating-point constants to the CIRCT frontend in https://github.com/llvm/circt/pull/7086. I think this is particularly interesting because we'd have to figure out how to represent these constants in Calyx. Verilog itself (AFAIK) does not support IEEE-754 constants and so we'd have to generate the right bit-patterns.

One possible approach is adding this capability to `calyx-frontend` but the IR only ever sees bit-patterns for the numbers. In this case `from_ast` becomes responsible for transforming a floating-point constant string into the correct bit-pattern.

ethanuppal, 2024-07-02:
@jiahanxie353 How's the progress on this?

jiahanxie353, 2024-07-02:
These two commits:
- [json and data parsing](https://github.com/calyxir/calyx/pull/1928/commits/c2e00e53cb02bba6d5aecf6a860a76510c34aeb6);
- [frontend parsing](https://github.com/calyxir/calyx/pull/1928/commits/86434a53fbba47b5525c91ad8d0316b0d71dd1b9)
supports the feature.

rachitnigam, 2024-07-03:
Thanks @jiahanxie353! Can you open a PR for these two commits so we can start the review?

jiahanxie353, 2024-07-03:
Sure, just created it!