calyxir/calyx #2066: [fud2] Mode that is friendly to parallel builds, with a unique per-run workdir name

sampsyo, 2024-05-27:
fud2 does all its work in a *working directory*. By default, it creates (and cleans up) a directory called `.fud2` in the user's current working directory. But you can also override it with `--dir` if you want to do the build anywhere.

The default setting is not suitable for Runt tests and other situations where multiple invocations of fud2 happen in parallel. The problem, of course, is that multiple builds try to happen in the same directory and clobber each other. So, we need a mode where different builds get a unique working directory, with an auto-generated name that is distinct from any other execution. We should of course still keep the current behavior in another mode, because (especially with `--keep`) it's nice to have a predictable workdir name for most builds.

There are two decisions to make here, one minor and one more significant:

1. Minor: In this mode, should the temporary directory still go in the user's current working directory (by default), or should it go in `/tmp` by default? Like, we could just call the newly-generated directory `.fud2-c2bf0f3c` and put it where we would otherwise normally put it. Or we could behave like most other temporary-directory-having commands and put it in `/tmp/fud2-c2bf0f3c` or whatever.
2. Major: Should this new behavior be the default? We could decide that most users, most of the time, would like (or wouldn't mind) a fresh directory name for most builds. Of course, this would ruin all incrementalism, but that's already true if you don't use `--keep`. So we could use fresh names by default and use the stable name `.fud2` only when using `--keep`.

sampsyo, 2024-05-28:
Awesome, thanks, @nathanielnrn. Incidentally, accidentally-remaining `.fud2` may be somewhat less of a problem as of #2058, but probably not gone altogether.

Sounds like @ethanuppal is interested in giving this a try, with the design you endorse as well.

ethanuppal, 2024-05-28:
Sorry it's taken me so long -- I'll be making a pull request from my fork shortly

ethanuppal, 2024-05-30:
@sampsyo, I don't think I'm the right person to make a PR yet, given that I'm not familiar with the organizational semantics of the codebase (e.g., my misplaced test) or comfortable enough to make "drastic changes" (I knew that there would be duplication of functionality when I made the temporary directory separately, but I didn't want to make large changes to a codebase I haven't touched yet, hence why I tried to implement the minimally invasive solution). It might be better for someone else to give it a try.

nathanielnrn, 2024-06-05:
Closed via #2092. Unique directory names are now the default when not using `--keep`

closed 2024-06-05 by nathanielnrn