calyxir/calyx #2067: Tracker: binary heaps

anshumanmohan, 2024-05-28:
#1999 motivates the need for a general PIFO. It also discusses a made-up strategy for implementing such a PIFO. It leaves on the table, however, another straightforward solution: binary heaps.

This is a tracker for the things we need.
- [ ] A module that accepts values and `push`es them into a min-heap ordered by value.
- [ ] Tweak the above to accept a value and a rank, and `push` a (value, rank) [tuple](https://github.com/calyxir/calyx/blob/513fda8bff119eea97e645b19dbbacf674f8a5a0/calyx-py/test/correctness/tuple.py) ordered by rank.
- [ ] Modify the above to obey the [queue interface](https://docs.calyxir.org/frontends/queues.html#shared-interface).
- [ ] Add `peek` and `pop` functionality.
- [ ] Test!

ethanuppal, 2024-05-29:
Does this issue have a PR or a branch?

anshumanmohan, 2024-05-29:
PR for the first task coming soon. My intention is to leave the other tasks as little tasks for my BURE undergrads.

anshumanmohan, 2024-05-29:
@ethanuppal see #2074 if you're curious!

closed 2025-02-04 by polybeandip