calyxir/calyx #2109: Question about Section 3 of the paper

xiuhu17, 2024-06-05:
As my knowledge, non-blocking assignment is ```always_ff``` in hdl(for example, SystemVerilog). The value from right hand side will not be shown in the left hand side immediately(the same cycle).
[image]

sampsyo, 2024-06-06:
Hi there! I'm glad you're taking a look at the paper!

Could you please give a little more detail about what your question is here exactly? Not sure if this is what you're looking for, but assignments in Calyx are a little bit like `always_comb` in SystemVerilog, in the sense that they take place within a clock cycle, but they are not ordered, which is not true of `always_comb` but is kinda more like `<=`. So any analogy to Verilog is necessarily imperfect.

xiuhu17, 2024-06-06:
Thanks for the reply!

I think you solved my confuse. It is stated as non-blocking since the assignment is not ordered, sth like ```always_ff```, except the assignment is done in the same cycle.

closed 2024-06-06 by rachitnigam