calyxir/calyx #2166: [calyx-py] Case statements proposal

nathanielnrn, 2024-06-19:
I think at the moment there is no concise way to express case statements in pure Calyx. As a result, the eDSL doesn't currently have a way to express these concisely either. 

A possibility for adding a `case` builder in the eDSL could be something like this:

`case` would be a [`ComponentBuilder`](https://github.com/calyxir/calyx/blob/main/calyx-py/calyx/builder.py#L67) method. The signature would be
```python
case(self, signal : ExprBuilder, cases : List[Tuple[int,As_Controllable]])
```
Where `As_Controllable` is anything that [`as_control`](https://github.com/calyxir/calyx/blob/main/calyx-py/calyx/builder.py#L1043) can be passed without failing. So things like invokes and groups, among others.

This would generate a control sequence using `if` statements in the `control` block of the component.

A use case would look like
```python
my_comp = prog.component("my_comp")
comp_reg = my_comp.reg(1, "comp_reg")
in_1 : ExprBuilder =  my_comp.input("in_1", 8)

with my_comp.group("my_group") as my_group:
    # Some assignments

my_invoke = invoke(comp_reg, in_in=1)
my_comp.case(in_1, [(1, my_group), (2, my_invoke)])
```

Would produce calyx that looks something like this

```
component my_comp (in_1 : 8) -> () {
  cells{
    comp_reg = std_reg(1);
    //Below here are things generated from the case statement
    case_eq_1 = std_eq(8); //width inferred from in_1
    case_eq_2 = std_eq(8);
  }

  wires{
    group my_group {
      // Some assignments
    }
    //Below here are generated from case statements
    case_eq_1.left = 8'b1;
    case_eq_1.right = in_1;
    case_eq_2.left = 8'b2;
    case_eq_2.right = in_1;
  }

  control{
    if case_eq_1.out {
      my_group;
    }
    if case_eq_2.out{
      invoke comp_reg(in=1'b1)l
    }
  }
}
```

A few questions that might be worth thinking about:

1. Instead of implementing `case` semantics in `control` blocks, we could have the case statement be expressed in a component's `wires` block with guarded assignments? If we have cases `a`, `b`, and `c` we could have
    ```
      case(signal):
        a : foo <= 2'b1;
        b: foo <= 2'b0;
        c: foo <= 2'b2;
    ```
    represented in Calyx as

    ```
    foo = a ? 2'b1;
    foo = !a & b ? 2'b0;
    foo = (!a & (!a & b) ? 2'b2;
    ```
    Where each assignment guard `and`s and inverts all of the cases that came before it.
    
    I imagine this might allow Verilog to better simplify these boolean expressions, as opposed to making our compiler perform a bunch of control optimizations? Especially because the compiler does not know that the `if` guards are mutually exclusive. The expansive guards does contain that information.

2. The way this is implemented above, with a bunch of `std_eq`s, doesn't seem to have a nice way to have a `default` case. We'd need to do a bitwise OR which afaik would require very deeply nested `std_or` trees or a bunch of wire manipulation that isn't easily expressible in pure Calyx. Are we okay with this? This seems like a pretty large flaw, but this should be enough for cases where we want to "do nothing" if no case matches our signal.

3. A first class `else if` in Calyx might make all of this a non-issue. Should we pursue something like that instead?

nathanielnrn, 2024-06-19:
cc @anshumanmohan

sampsyo, 2024-06-19:
Sounds great! One very low-level Pythony suggestion is that, instead of using a list of pairs, we could use a dict. (Dicts are ordered in Python.) So this syntax might look nice:

```py
my_comp.case(in_1, {
  1: my_group,
  2: my_invoke,
})
```

sampsyo, 2024-06-19:
And, about the "control" style or the "guarded assignment cascade" style: I think these probably serve different functions, and there is even a role for both to co-exist if we want. That is, the control style supports arbitrary blocks of control, and the second style just supports assignments of ports (or constants). Maybe it's worth deciding which is most pressing for your use case and prioritizing that?

You're right that a "default" might be hard to do, at least in a first cut. I might recommend forging ahead without support for default cases and coming back to that later.

`else if` in the IL is certainly a possibility, but I wouldn't let its absence hold back doing something nice on the builder side! The Python library could even be evolved to target the new control construct if it becomes available.

nathanielnrn, 2024-06-19:
Ah right, I was conflating the two because the particular use case motivating this is constrained to very few/single assignments in each group. I think the control style will interface a bit nicer with the rest of the builder, so going to move forward with that for now.

anshumanmohan, 2024-06-19:
I like it a lot! The queues are currently doing a discount version of this (branching one of 4 ways based on a 2-bit command signal). See [here](https://github.com/calyxir/calyx/blob/914a6111fc29ce9bd0a0263a8c717d5e8817dbc8/calyx-py/test/correctness/queues/fifo.py#L71-L79) for FIFO's control block.

nathanielnrn, 2024-06-20:
Right! Thanks for the input everyone. Wrapping this in a `par` is the [plan](https://github.com/calyxir/calyx/pull/2168/files#diff-1b6ee22a9e8b557927fd938de3186a9db39b2efd86dd63605f498edeb3d75160R231).

closed 2024-06-21 by nathanielnrn