calyxir/calyx #2187: Explore what it would take to implement PIFO trees using these binary-heap PIFOs.

anshumanmohan, 2024-07-01:


anshumanmohan, 2024-07-22:
@KabirSamsi just a quick request to please do a little documentation in this thread to capture our discussion from today! Don't spend too much time on it, but do get down the basic concrete plan so that we can look at it together. I think it'll really help you when you get back to this in a few days :)

anshumanmohan, 2024-07-23:
Actually @KabirSamsi let's try and meet tomorrow morning if that's possible. Like, before you put in any energy towards a write-up. On further thinking I may have given you a bit of #fakenews during our chat today. The thing I'd like to revisit is your starting point. I told you you'd want to start with Calyx and create a Calyx-level tree, but I now think you may be able to start with OCaml and create a Calyx-level tree. @polybeandip feel free to join this chat, since it will involve a handshake with your work

KabirSamsi, 2024-07-23:
Yes sounds good! To both :)

PR #2399 (2025-02-04): Queues: implement PIFO tree with binary heaps
Closes #2187!

Changes:
- add [`tree.py`](https://github.com/calyxir/calyx/blob/binary-heap-tree/frontends/queues/queues/tree.py), a wrapper for building "tree queues" given the components for the root and children
- use `tree.py` to implement [`binheap/complex_tree_test.py`](https://github.com/calyxir/calyx/blob/binary-heap-tree/frontends/queues/tests/binheap/complex_tree_test.py), a binary heap version of [`complex_tree_test.py`](https://github.com/calyxir/calyx/blob/binary-heap-tree/frontends/queues/tests/complex_tree_test.py)
commit 7925f23b (2025-02-04): Queues: implement PIFO tree with binary heaps (#2399)

Closes #2187!

Changes:
- add
[`tree.py`](https://github.com/calyxir/calyx/blob/binary-heap-tree/frontends/queues/queues/tree.py),
a wrapper for building "tree queues" given the components for the root
and children
- use `tree.py` to implement
[`binheap/complex_tree_test.py`](https://github.com/calyxir/calyx/blob/binary-heap-tree/frontends/queues/tests/binheap/complex_tree_test.py),
a binary heap version of
[`complex_tree_test.py`](https://github.com/calyxir/calyx/blob/binary-heap-tree/frontends/queues/tests/complex_tree_test.py)

-------- [... cut]
closed 2025-02-04 by polybeandip