calyxir/calyx #2191: Queues: flag day

anshumanmohan, 2024-07-02:
We have been zooming along and making quite a few improvements to our queues, and we have decided on a small clutch of TODOs for the magical day when things settle down. This is a tracker for all that stuff. It will probably need to grow further; feel free to edit this topmatter directly to add to the list.

- [ ] https://github.com/calyxir/calyx/issues/2186
- [ ] https://github.com/calyxir/calyx/issues/2170
- [ ] Neaten the code all around. Use [`case`](https://github.com/calyxir/calyx/pull/2168) to simplify our "menus". Simplify the logic when possible.
- [ ] Consider a test harness that spares us from checking in large `.data` and `.expect` files. We would probably still need to _make_ these files or something like them, but perhaps we can make them on the fly and then discard/`.gitignore` them. Super open to ideas on this one. @ethanuppal seems to have some thoughts.
- [ ] Beef up the [documentation](https://docs.calyxir.org/frontends/queues.html) as needed.
- [ ] More!

ethanuppal, 2024-07-07:
The idea with testing harnesses is that you don't need to have any files. You can call the calyx component as a function, so you can generate a random large array, enqueue those values, and dequeue them — all from code!

csziklai, 2024-07-24:
I will do the documentation for round robin and strict PIFOs.

polybeandip, 2024-07-27:
As part flag day, do we want to remove `peek` from the [queue interface](https://docs.calyxir.org/frontends/queues.html) and make `cmd` a 1-bit signal?

anshumanmohan, 2024-07-28:
Good question, I’ve kinda been wanting to. The reason to keep it is that we may find an algorithm that needs it. The reason to lose it is that we haven’t yet found such an algorithm. Let’s make a call about it on Monday!

KabirSamsi, 2024-07-29:
As of 7/29:

- We are removing the `peek` functionality from all queues, and the `pop/peek` by value functionality for PIEOs.
- Subsequently, we will be changing `cmd` to a 1-bit signal.
- We'll proceed with flag day before the Calyx PIEOs are merged, and continue work with that afterwards.

anshumanmohan, 2024-08-09:
I've checked the documentation box manually; let's just declare the docs good!

closed 2025-02-05 by polybeandip