calyxir/calyx #2208: [Fud2] Cider is the default for `--to dat` computations

EclecticGriffin, 2024-07-15:
It should probably be verilator for the sake of matching the original `fud` and for overall stability

sgpthomas, 2024-07-16:
I think defaults are decided based on the order that they are added. I think that the right solution to this problem is being able to explicitly declare priorities of operators. So that it doesn't depend on something rather opaque.

EclecticGriffin, 2024-07-16:
Yeah I agree with that approach

sampsyo, 2024-07-30:
We definitely need a way to do something about this so that the default paths are more obvious & predictable. The tracker (#1878) mentions priorities as a way about this, and it links to #1873 for further discussion. I will link the tracker to this issue as well as a specific instance of the prioritization problem, which we should perhaps fix even if we don't have a general scheme yet.

In general, however, I'm not 100% sure integer priority scores are the way… they are a little bit brittle and fiddly to manage. Maybe there's another way that we can ensure a deterministic order of op registration (including from Rhai files) that would allow this kind of control without scoring.

rachitnigam, 2024-07-30:
> Maybe there's another way that we can ensure a deterministic order of op registration (including from Rhai files) that would allow this kind of control without scoring.

The downside is that this will require fiddling with Rhai for users who only care about using `fud2` (instead of adding new stages, ops, etc.)

sampsyo, 2024-07-30:
I think I understand what you're saying: the the _user_ should also be able to specify what to prioritize. That wasn't what I was thinking of here; it was just that the fud2 _developer_ should have a way to predictably induce the default that they want.

Separately, yeah, I guess it could also make sense to let users pick whether they want to behave like `--through icarus` or `--through verilator` as the default behavior, just by modifying their config file. But that seems like a somewhat more niche thing; it's more important that the _default_ defaults be controllable by fud2 developers.

rachitnigam, 2024-10-26:
Is this issue still relevant given that the `fud2` tracker is also tracking the issue? If it is, let's change the name to something that's more specific to the problem that we want to fix.

EclecticGriffin, 2024-10-28:
I guess we can close this in favor of the general priority issue

closed 2024-10-30 by EclecticGriffin