calyxir/calyx #2241: Queues: make `cmd` a 1-bit signal by removing `peek` functionality from all queues

polybeandip, 2024-07-31:


KabirSamsi, 2024-08-01:
To further add to this, a specialization between Work-Conserving and Non-Work-Conserving Queues:

**WC**: Make `cmd` a 1-bit signal and remove the `peek` command – hence, it will flip between `push` and `pop`.
**NWC**: Make `cmd` a 1-bit signal and remove the `peek by value`, `peek by time` and `pop by time` commands - hence, we will flip between `push` and `pop by time`.

anshumanmohan, 2024-08-01:
Hint for `fud` commands that give your resource estimates:
- https://github.com/cucapra/calyx-resource-eval/blob/main/sdn/scripts/sdn_four_ways_latency.sh
- https://github.com/cucapra/calyx-resource-eval/blob/main/sdn/scripts/sdn_four_ways_resources.sh

calebmkim, 2024-08-01:
> What is the commit hash, and where can I find it?

The commit hash is a unique identifier created by git for each commit to a branch. You can think of the commit hash as specifying which version of the Calyx compiler you want to build and compile your designs with. Most of the recent commits should produce the same/similar code for a given design, so I don't think the commit matters too much, you can probably just pick a relatively recent one. 

If you click on "2,875 commits" button, you can see a full commit history with the hashes. 
[image]
[image]

> Also what are the params, as in what do they mean and what are they parameters to?

tl;dr you can probably just leave it empty. 

In Calyx, you can pass parameters to the compiler that gives specific information on how the compiler shold run. For example, there is a pass called `static-promotion` which converts dynamic Calyx code to static code. If you wanted to delete this pass you would do `-d static-promotion` to the Calyx compiler. Or, if you want to promote code, but not compact (i.e., automatically parallelize certain parts of the code during promotion) code then you would pass `-x static-promotion:compaction=false`. In the `params` this would correspond to "[static-promotion,compaction,false]". If you just want to run the default pipeline, then you can leave the field empty.

commit 369e6843 (2024-08-08): Remove peek functionality (#2243)

This PR closes #2241. It removes peek from all the queues, making `cmd`
now a 1-bit signal. Please note that the corresponding update in the
documentation `queues.md` is part of #2237.

One minor issue is that now that peek is removed, round robin no longer
uses the group `restore_hot`, leading to unused group error. Strict
still uses `restore_hot` so I can’t delete it, and if you remember
gen_strict_or_rr.py is used to make both strict and round robin queues.
I’ve gone ahead and turned off wellformedness in the runt file as a
temporary solution because I’m n [... cut]
PR #2243 (2024-08-08): Remove peek functionality
This PR closes #2241. It removes peek from all the queues, making `cmd` now a 1-bit signal. Please note that the corresponding update in the documentation `queues.md` is part of #2237.

One minor issue is that now that peek is removed, round robin no longer uses the group `restore_hot`, leading to unused group error. Strict still uses `restore_hot` so I can’t delete it, and if you remember gen_strict_or_rr.py is used to make both strict and round robin queues. I’ve gone ahead and turned off wellformedness in the runt file as a temporary solution because I’m not sure w [... cut]
closed 2024-08-08 by csziklai