calyxir/calyx #2251: Require that Python code conform to the Black formatter

anshumanmohan, 2024-08-06:
Let's expand the scope of mandatory Black formatting in CI. [Currently](https://github.com/calyxir/calyx/blob/894525448d0269f044154d298c1216ea41d1e2c3/.github/workflows/format.yml#L42-L53) only two directories are tested, but I think this can safely be expanded to include all of `calyx-py` for instance.

Right now if I run `black calyx-py` I get
```
am3327@havarti:/scratch/anshuman/calyx$ black calyx-py/
...
All done! ✨ 🍰 ✨
22 files reformatted, 32 files left unchanged.
```

@polybeandip interested? Since it bit you once before and you figured this out in OCaml-land recently?

polybeandip, 2024-08-06:
Happy to set up the github action for this

polybeandip, 2024-10-04:
@anshumanmohan are you sure that your link to the action in `format.yml` actually checks that the python files in `fud/` and `frontends/systolic-lang` have been formatted?

From the [docs](https://black.readthedocs.io/en/stable/integrations/github_actions.html):
> Please note that the [--check flag](https://black.readthedocs.io/en/stable/usage_and_configuration/the_basics.html#labels-exit-code) is required so that the workflow fails if Black finds files that need to be formatted.

ATM, none of our current calls to `psf/black@stable` actually have the `--check` option. You can also inspect previous runs of CI to see that both these directories have files that black would have formatted, but CI doesn't fail.

Shall I setup CI so that it fails if black would format any files in `fud/`, `frontends/systolic-lang/`, `frontend/queues/`, or `calyx-py`? I imagine I should probably ask the people maintaining `fud` and `systolic-lang` before doing that.

anshumanmohan, 2024-10-04:
Thanks for the catch! I just operated in faith and assumed that it was in fact working. But, as you've noted, there seem to be some slippages. To me it feels 100% worth doing, but let's get green lights from @calebmkim and @rachitnigam anyway!

calebmkim, 2024-10-04:
Sounds good to me!

PR #2295 (2024-10-06): Setup CI formatting check for Calyx-Py and Queues
Closes #2251 

Changes:
- Run `black --line-length 88` on `frontend/queues`, `calyx-py`, and `fud`
- Tweak `.github/workflows/format.yml` to check formatting for these directories
    - add the `--check` options to all calls to `black`
commit 7260196e (2024-10-06): Setup CI formatting check for Calyx-Py and Queues (#2295)

Closes #2251 

Changes:
- Run `black --line-length 88` on `frontend/queues`, `calyx-py`, and
`fud`
- Tweak `.github/workflows/format.yml` to check formatting for these
directories
    - add the `--check` options to all calls to `black`
closed 2024-10-06 by rachitnigam