calyxir/calyx #2263: `json-dat.py` script is hard to install with new pip restrictions

sgpthomas, 2024-08-14:
The new `pip` restrictions force you to install everything in a virtualenv. This makes it very annoying to run fud/fud2 commands that rely on this without resorting to disabling this error, or making new virtualenvs. Both of these seem non-ideal. I think probably the solution is to turn `json-dat.py` into an installable python package, but I'm open to ideas. The other reasonable possibility that I can imagine is to remove all python dependencies except the fud dependency. And then set `PYTHONPATH` to include the `fud` directory when we run `json-dat.py`.

sgpthomas, 2024-08-15:
I think a better solution is to set the `python` var in `fud2.toml` to a venv python with the correct things installed. This is putting some of the burden on plugin writers to not use python directly, and instead read the python version from the config. Which I think most of them do already.

rachitnigam, 2025-02-03:
#2404 might help solve this

EclecticGriffin, 2025-06-30:
#2480 automatically sets the `python` var in the fud2 configs to an appropriate venv. So I believe this is also resolved