calyxir/calyx #2308: Primitives aren't included in file output

bcarlet, 2024-10-19:
When using the Verilog backend, the primitives aren't included in the output when writing to a file rather than standard out. This is because the file is truncated when it's opened a second time in the `OutputFile::get_write` call in `VerilogBackend::emit`, which overwrites anything that was written by `VerilogBackend::link_externs`.

I'll fix this, but I'm recording it in an issue since I probably won't get to it immediately.

rachitnigam, 2024-10-21:
Thanks @bcarlet! Can you provide an MRE?

bcarlet, 2024-10-21:
Sure, take any file that imports primitives:
```
import "primitives/compile.futil";

// The component doesn't matter
component main() -> (out: 8) {
  cells { r = std_reg(8); }
  wires { out = r.out; }
  control { invoke r(in=8'd42)(); }
}
```

If we we compile it to Verilog and send the result to standard out,
```
calyx -b verilog example.futil > out.sv
```
then `out.sv` contains a `module std_reg`. But if we send the result directly to a file,
```
calyx -b verilog example.futil -o out.sv
```
then `out.sv` does not contain a `module std_reg`.

PR #2317 (2024-10-26): OutputFile calls create just once per run
Fixes #2308
commit c6c92ba0 (2024-10-26): OutputFile calls create just once per run (#2317)

Fixes #2308
closed 2024-10-26 by rachitnigam