calyxir/calyx #2321: Prevent Verilog keywords from being used as names

polybeandip, 2024-10-29:
Here's an example program, where we name a cell `wait`:
```
# making a cell named "wait"
import "primitives/core.futil";

component main() -> () {
  cells {
    wait = std_reg(32);
    add = std_add(32);
  }
  wires {
   static<1> group write {
     add.left = wait.out;
     add.right = 32'd1;
     wait.write_en = 1'b1;
     wait.in = add.out;
   }
  }
  control {
      write;
  }
}
```
Running this with `fud exec bad.futil --to dat --through verilog` yields
```
[fud] ERROR: `/opt/verilator/bin/verilator --trace /tmp/tmpcsy0je3b /scratch/ad739/calyx/fud/icarus/tb.sv --binary --top-module toplevel --Mdir /tmp/tmpq176rkir -fno-inline>&2' failed:
=====STDERR=====
%Error: /tmp/tmpcsy0je3b:355:3: syntax error, unexpected wait, expecting IDENTIFIER or randomize
  355 | ) wait (
      |   ^~~~
%Error: Exiting due to 1 error(s)

=====STDOUT=====
```
Perhaps this kind of code should give the user a Calyx error instead of a Verilator one? This way the Calyx compiler isn't producing malformed Verilog code.

polybeandip, 2024-10-29:
Oops, looks like there's already a file that handles these issues. Adding `wait`  (and, another one I found, `break`) to [`reserved_names.rs`](https://github.com/calyxir/calyx/blob/e1641e6b03c653f3f3659aacbf2f6c275f0d44b6/calyx-ir/src/reserved_names.rs#L2) should do the trick

rachitnigam, 2024-10-30:
@polybeandip can you link the PR to this issue when you have one? I'm going to mark the issue stale in case there is no activity on this for a bit

PR #2322 (2024-10-30): Add `wait` and `break` to reserved names
Closes #2321
commit b0849747 (2024-10-30): Add `wait` and `break` to reserved names (#2322)

Closes #2321
closed 2024-10-30 by polybeandip