calyxir/calyx #2336: Floating-point conversion error

ekiwi, 2024-11-05:
There is a bug in the tests provided by https://github.com/calyxir/calyx/pull/2318.

[tests/correctness/ieee754-float/parse-and-print.futil.data](https://github.com/calyxir/calyx/pull/2318/files#diff-e4146a309a9876672f57eb53e44b7c9042a5a222c53eb6894ff48df0c6a950fd) provides the input value `15.1239`. However, at the end of the test, the value has changed to `15.12` in [tests/correctness/ieee754-float/parse-and-print.expect](https://github.com/calyxir/calyx/pull/2318/files#diff-5d85d972c197f71bd7796802102b6e470a03f2b8cf64f277a0f8208dffcaca99).

This is even though a 32-bit IEEE float should be able to represent `15.1239`. So there appears to be a bug, most likely in how the data is converted back after the simulation.

rachitnigam, 2024-11-05:
Argh, thanks for catching this! Also tagging @jiahanxie353 in case he can get to this before me.

EclecticGriffin, 2024-11-05:
Probably an easy fix? Best guess is that the problem comes from this line:
https://github.com/calyxir/calyx/blob/6b4896976f93f7a913fed46de064f17bc5468b10/fud/fud/stages/verilator/json_to_dat.py#L45

PR #2337 (2024-11-06): Disable rounding for float conversion
Fixes #2336. Currently running into the same problem as https://github.com/calyxir/calyx/pull/2335#issue-2636162273

@ekiwi how did you fix it on the Rust side?
commit f44cf8ff (2024-11-06): Disable rounding for float conversion (#2337)

Fixes #2336. Currently running into the same problem as
https://github.com/calyxir/calyx/pull/2335#issue-2636162273

@ekiwi how did you fix it on the Rust side?