calyxir/calyx #2348: `fud2` silently fails with errors

rachitnigam, 2024-11-12:
When adding a new stage, `fud2` will silently fail when there is a problem. For example, I am adding a new Dahlia stage and when the key is absent, instead of giving an error, `fud2` generates an incorrect script:

```
import "calyx" as c;

export const dahlia = state("dahlia", ["fuse"]);

defop dahlia_to_calyx(prog: dahlia) >> out: c::calyx_state {
    let dahlia = config_or("dahlia.exec", "dahlia");
    let flags = config_or("dahlia.flags", "");

    shell(`${dahlia} ${flags} ${prog} > ${out}`);
}
```

With the command:
```
fud2 tests/frontend/dahlia/binop_tree.fuse --to calyx -m emit
```

Will generate:
```
build-tool = /Users/rachitnigam/.cargo/bin/fud2
rule get-rsrc
  command = $build-tool get-rsrc $out

# dahlia_setup (plugin)
Failed to load plugin: dahlia-to-calyx.rhai
  Emitter error: missing required config key: dahlia

# build targets
build _to_stdout_calyx.futil: dahlia-to-calyx tests/frontend/dahlia/binop_tree.fuse

default _to_stdout_calyx.futil
```

This should be considered a blocking usability issue for the `fud2` transition.

EclecticGriffin, 2025-04-14:
So I've got a proof of concept kludge that changes the output to:
```
> fud2 tests/frontend/dahlia/binop_tree.fuse --to calyx -m emit

build-tool = /Users/griffin/.local/bin/fud2
rule get-rsrc
  command = $build-tool get-rsrc $out

# dahlia_setup (plugin)
Error: Emitter error: missing required config key: dahlia
```

And importantly, exits with an error code. If we're satisfied with that, this should be an easy fix?
```
> fud2 tests/frontend/dahlia/binop_tree.fuse --to calyx -m emit > /dev/null

Error: Emitter error: missing required config key: dahlia
```

EclecticGriffin, 2025-04-14:
Not to step on @jku20's toes, of course! I can create a PR if this would be of interest

jku20, 2025-04-14:
I lay dormant for at least another month, so this I don't think that would be an issue.

EclecticGriffin, 2025-04-22:
Closing this for the moment since #2447 solves this, albeit inelegantly

closed 2025-04-22 by EclecticGriffin